LaserWeb/LaserWeb4 #348: Ability to move measurements pop-up

KirAsh4, 2017-06-16:
I often find myself selecting individual elements within a file to set a specific operation, be it cutting, engraving, or a fill path. With some of my designs, those elements are small and tight together. When I select one, often the measurement box that pops up will cover up others that I need to select as well. Can that pop-up box be made so it's moveable, and not forced in an absolute location related to the element(s) selected?

jorgerobles, 2017-06-16:
There's another issue asking to be fixed...

cojarbi, 2017-06-16:
Jorge this is one of the reason I was suggesting moving this to the top of the window but fixed like many other app do.

jorgerobles, 2017-06-16:
Yes :)

KirAsh4, 2017-06-16:
Yeah, top of the window would make more sense, there's a lot going on at the bottom already. Some of my CAD software has that on the bottom of the screen where you also enter console commands (AutoCAD for example.)

tbfleming, 2017-06-16:
I prefer movable over fixed so I can keep it near the objects I'm working on.

KirAsh4, 2017-06-16:
Either one would work. I don't often use any of the data in that box (except when I'm moving an entire design over on the material sheet), so if I have to look "up" at the top of the screen the few times that I do need it, I'm okay with that. Or having the ability to move it out of the way also works.

cojarbi, 2017-06-16:
I do agree having it near the object is less travel and convenient but how much interaction should the user have with the object to take advantage of that?

cojarbi, 2017-06-16:
Or maybe make it an option to make it "not sticky " to the selection

KirAsh4, 2017-06-16:
Right, it's there, it provides information, but I rarely actually use it or change the values, *except* when I'm moving design to a different spot on the material and I don't want to re-export it from the native application. I don't mind it being there, as long as I can move it out of the way. Or, stick it some place permanent (though I agree with the travel time.)

cprezzi, 2017-06-18:
I do very often use the box to change scale or reposition single objects ;)

jorgerobles, 2017-06-22:
Better handling! Now, the behavior is the same as always until you drag the floating from the crosses icon. Once moved, it will keep there until double click the crosses icon.

closed 2017-06-22 by jorgerobles

cojarbi, 2017-06-22:
Is it possible to make the change persistant while closing and opening the app :)