LaserWeb/LaserWeb4 #351: AI SVG text not showing up correctly

cojarbi, 2017-06-19:
Could someone verify why the letters ( converted to outlines ) dont show up correctly?
Made on AI CC 2017
[LaserWeb Logo copy.svg.txt](https://github.com/LaserWeb/LaserWeb4/files/1086607/LaserWeb.Logo.copy.svg.txt)
[image]

cojarbi, 2017-06-19:
Update: its seems its not recognizing the fill. If i add a stroke to the letter then it shows up

tbfleming, 2017-06-20:
The SVG doesn't have any fill or stroke set.

cojarbi, 2017-06-20:
In AI it does have a fill

⁣Ariel Yahni​

jorgerobles, 2017-06-20:
On export, use representation attributes instead css. Black fills are not
correctly exported by ai

cojarbi, 2017-06-20:
@jorgerobles not working for me. I think i have tried all the possible combinations. 

This other file was just made in inkscape, text converted to path and added fill / stroke. LW does not load it at all and says " its not your fault " which  i like
[drawing.svg.txt](https://github.com/LaserWeb/LaserWeb4/files/1088031/drawing.svg.txt)

tbfleming, 2017-06-20:
I just pushed a fix to the Inkscape problem.

tbfleming, 2017-06-20:
It had code to handle the case where the fill and stroke were missing. It created a transparency effect to indicate the problem, without annoying people who intentionally made things invisible. The recent frame buffer caching system caused it to be too transparent. fixed.

cojarbi, 2017-06-20:
Woking perfectly. Thanks @tbfleming

closed 2017-06-20 by cojarbi