LaserWeb/LaserWeb4 #354: [:unicorn:] Optical Mark recognition

jorgerobles, 2017-06-24:
Yeah, we have the unicorns back!
What about this kind of thing https://github.com/jcmellado/js-aruco to jog detect work start, I mean, precisely warn of material placement in order to cut for example, previously printed cardboard or so?

jorgerobles, 2017-06-26:
@openhardwarecoza Other way to implement a placement tracker could be with a reflectance sensor matrix https://es.aliexpress.com/store/product/10PCS-lot-QRE1113-new-in-stock/125218_32707304995.html --- maybe an i2c board with a multiplexer?

openhardwarecoza, 2017-06-26:
> QRE1113-new-in-stock/125218_32707304995.html --- maybe an i2c board with

Personally, i'd go for a camera approach, since it doesn't involve licking
anyone to get firmware support for the IR (;. More under our control

arthurwolf, 2017-06-26:
FYI, as part of the smoothie project, we have a sub-project we are working
on that is related to that feature ( not named yet )
it's essentially a program that would run on a raspi ( or raspi zero ), and
that offers a web server ( ajax api ) to allow other programs ( like
laserweb, visicut or fabrica ) to take pictures ( via one or more webcam
plugged into the raspi ) and have opencv operations done onto them ( like
stitching of mosaics, reformation of fisheye pictures, corner/edge finding
). the whole assembly would be on the head of the fabrication machine.
we are actually paying a consultant to work on this project and plan to
release it both as a fully open-source project, and as a ready-to-use
product

the final product would include :
* a raspi zero for the brains
* a 200x usb microscope for precise edge finding
* a fisheye camera to take quick global images of the work area to find
what's on it
* a laser to help with edge finding and distance measurement ( height to
plate )
* a time-of-flight distance sensor to help with fast work area height
scanning
* one IR and one UV spectrographic sensor, which hopefully will allow us to
detect spectra of material and auto-detect if something is wood or plastic
for example
* an arm and a motor to auto-retract from the sensing position when not in
use

users would then be able to just install this on the head, tell their
favorite host program what it's ip is, and that host program would then be
able to use it to :
* capture ( including a z-height map ) what's on the work area and show it
to the user, allowing the user to align imported files with the actual
workpiece. it's not just a network camera, it actually moves the head at
several points on the work area and joins the pictures together to make a
single work area picture
* precisely find the corner of things ( very important for cnc ) with very
minimal user effort
* produce 3D scans of what's on the work area, including "simplification to
basic solids" for better UI representation of the workpiece
* other stuff we need to test a bit more before talking about publicly
the system would do most of the thinking ( using opencv mostly ) and would
leave the host with as simple and as easy as possible an interface, so the
system would be pretty easy to integrate ( just add a bunch of modals that
pop up at the right moments, do a few ajax calls and display a few pictures
)

just thought I'd mention this as you guys seem to be thinking about similar
problems, and maybe knowing this is in the works might have some importance

jorgerobles, 2017-06-26:
That's a really good news. And seems a full blown contraption :)

jorgerobles, 2017-06-26:
http://recordit.co/JmShd46wkk ... not bad being JS

openhardwarecoza, 2017-06-26:
@jorgerobles recordit doesnt do it justice!   Low FPS because of recordit. Trying it live with own camera - i was stunned.  That multimarker demo realtime recognises over 40 targets!

jorgerobles, 2017-06-26:
Yes! The Earth demo (estimating position) is slower, but I think could work decently. I need to make a test with a shape, and check deviation of a manual aligned work. If that works well enough, could be gold. at least with diode cutter. A perfect setup could rotate all the artwork on LW to match paper sign. :D

openhardwarecoza, 2017-06-27:
https://trackingjs.com (eg https://trackingjs.com/examples/color_camera.html ) is also nice and fast.  You gave me a new bug:  computer vision was always out of reach, but now thats its JS its up my skill level at last!  Dont know what i want to do with it, but i know i want to use it for something

jorgerobles, 2017-06-27:
> examples/color_camera.html ) is also nice and fast. You gave me a new

😈

jorgerobles, 2017-06-30:
I've done some tests of alignment with JS aruco and are promising
My setup:

[image]

[image]

Actually the tests are very basic:
- Positioned camera over the Marker. 
- Moving material until get an acceptable position (x0, y0, pitch roll etc)
[image]

- Several tests to calculate the camera offsets (Couple of them found -20,3mm in my setup)
- Run!

[image]

Caveats:
- Due to my magnificent $4 camera the marker estimation is not optimum but enough!
- Have to shift workspace via G92. If implemented, better offset method will be done.
- The addhoc pen holder does not mantains pen squareness

But hey, I've got no Unicorn, but a stubborn Donkey!

jorgerobles, 2017-07-02:
[image]

jorgerobles, 2017-07-02:
So, trying to not to derail UI even more, and making this useful, need some advices.

I've planned to add these settings on camera settings folder:

- Use OMR (toggle): toggles image postprocessing on/off on camera feed (CPU intensive)
    1. marker size (mm)
    2. offset [x,y]: will affect the gcode generation offset.
    3. marker generator? popups an svg with a random marker. Optional.

Intended usage:
 - setup your machine with a camera and a somewhat known offset 
 - search for zero manually, jogging until estimator goes almost zero on all values (depends on setup precision, of course)
 - set zero.
 - generated gcode will shift the coordinates the settings OMR offset if active.

What do you think?

tbfleming, 2017-07-02:
> offset [x,y]: will affect the gcode generation offset.

I'm uncomfortable with this. I thought mark recognition was to aid setting zero.

jorgerobles, 2017-07-02:
@tbfleming  Should be better run a G92 X{-xoffset} Y{-yoffset}?

tbfleming, 2017-07-02:
I suspect G92 may cause confusion. @cprezzi ?

jorgerobles, 2017-07-02:
Some sci-fi could include applying detected transform to the document :smiley:

tbfleming, 2017-07-02:
Hmmm. Do you plan on doing rotate? Unfortunately grbl can't do that using offsets. HAAS can, but that's a bit out of reach of our users...

jorgerobles, 2017-07-02:
Well my approach would be rotate the document on LW. Simplest, maybe. But is not a must.
First things first, It will suffice to do the zero offset.

tbfleming, 2017-07-02:
Since you're going to do rotate, might as well handle it in cam. Maybe a transform2d argument passed into preflight.

tbfleming, 2017-07-02:
1 marker might not be enough. e.g.:

* Assume only a 1 degree error detecting rotation (I suspect it will be worse)
* You're cutting a 100mm by 100mm rectangle
* Lower-left corner of rectangle is perfectly aligned
* Lower-right corner's y value will be 1.7mm off

jorgerobles, 2017-07-02:
Yes, I foresaw that issue. That's why on first instance rotation will be
skipped.
A 2/3 point registration could be done. At least registered with current
position, like reprap manual registration.

openhardwarecoza, 2017-07-02:
If the goal is stock rotation, why not use the built in xyz probing already
in all firmwares, using that to calculate rotation.

jorgerobles, 2017-07-02:
Well. My primary objective is/was have a decent registration point to make
cardboard models :) i didn't make my mind to other applications so far.

> Yes, I foresaw that issue. That's why on first instance rotation will be
> skipped.
> A 2/3 point registration could be done. At least registered with current
> position, like reprap manual registration.
>
>
> > 1 marker might not be enough. e.g.:
> >
> > - Assume only a 1 degree error detecting rotation (I suspect it will
> > be worse)
> > - You're cutting a 100mm by 100mm rectangle
> > - Lower-left corner of rectangle is perfectly aligned
> > - Lower-right corner's y value will be 1.7mm off

> >
> > <https://github.com/LaserWeb/LaserWeb4/issues/354#issuecomment-312507671

> >,
> BvZxy2LCakZO1ks5sJ91lgaJpZM4OEXzH>

JhhAeKNRj80XtzGNOWv2sDRJks5sJ999gaJpZM4OEXzH>

jorgerobles, 2017-07-02:
Seems G10 L2 should be used, http://linuxcnc.org/docs/2.6/html/gcode/gcode.html#sec:G10-L10 isn't it?

openhardwarecoza, 2017-07-02:
> Seems G10 L2 should be used, http://linuxcnc.org/docs/2.6/
> html/gcode/gcode.html#sec:G10-L10 isn't it?

Yes. Refer to existing set zero implementation

jorgerobles, 2017-07-02:
Anyways means that have to be set/reset on job start/end

cprezzi, 2017-07-02:
There is a big difference between `G10 L2` and `G92` (and not every firmware handles it the same way): 

- `G10 L2 P1` is setting **the offsets** only for G54 coordinate system to the given values.
- `G92` is setting **the actual position** to given values and is shifting the offsets of all coordinate systems accordingly.
- The frontend should not send these commands directly. Instead, the setZero command should be sent to the backend, which executes firmware specific commands.

cprezzi, 2017-07-02:
@jorgerobles How about a button "find mark" on the jog tab, that searches for the mark, moves to the calculated zero and sends setZero to the backend?

jorgerobles, 2017-07-02:
Yes, in order to affect offset, as camera is placed away from tool, could
you add some command to set it? :)

jorgerobles, 2017-07-02:
Overlapped answer! 
I think search for the code is overly complicated. I bet for jog to the marker, and once adjusted, click a set marker button. Also, as said before, the marker is not really zero. Well so I imagined, maybe I'm wrong

cprezzi, 2017-07-02:
I could add a setOffset command, if that helps.

cprezzi, 2017-07-02:
This way you could just call setPosition with the wished position that is calculated from marker position and camera offset.

jorgerobles, 2017-07-02:
Cool!

But should the offset be restored upon job finish? If so some remember
mechanism should be used, on frontend or directly in hardware with G10 Pn ?

cprezzi, 2017-07-02:
I wouldn't restore the previous value. The process is equal to jogging to the stock edge and pressing set zero. This could also not be reversed.

jorgerobles, 2017-07-02:
Ok then :)

cprezzi, 2017-07-02:
Implemented in https://github.com/LaserWeb/LaserWeb4/commit/867e5230384f19970810dbc3fb1781b17477ebfd

The param has to be an object with x, y, z, a (each optional). For example `setPosition({x:20, y:10});`

jorgerobles, 2017-07-03:
https://es.aliexpress.com/store/product/DBPOWER-USB-Endoscope-2MP-HD-IPX67-Inspection-Mini-Camera-15-10-7-5M-8-5MM-Lens/334521_32790703909.html  2Mp... Yumm

jorgerobles, 2017-07-04:
OMR branch alive https://github.com/LaserWeb/LaserWeb4/tree/OMR

jorgerobles, 2017-07-09:
There's also an scriptem artoolkit port. Very interesting.
I've installed opencv for python to compare performance and precision. No very different for me right now

tbfleming, 2017-07-09:
Nice!

jorgerobles, 2017-07-09:
JS Artoolkit precision seems far superior 🤤 
Now I have to figure how to use it :D

tbfleming, 2017-07-09:
If you get that building, then you'll also be able to build web-cam-cpp-src :)

jorgerobles, 2017-07-09:
Well I was planning using prebuilt js... Do I need to build?

tbfleming, 2017-07-09:
If you need to modify it

jorgerobles, 2017-07-11:
Before moving to other engine or get lost on other issues, does anyone get the itch and want to test?

cojarbi, 2017-07-11:
I ordered a camera. It will arrive in about months

jorgerobles, 2017-07-11:
LOL!

jorgerobles, 2017-08-14:
@tbfleming Looking into Artoolkit as an upgrade to Aruco, I see it return a float[16] matrix (http://augmentmy.world/artoolkit-distance-between-camera-and-marker)
I understand that matrix is relative to marker size in order to get a real measure.

Could you help me with the maths to get the Yaw,Pitch,Roll and dimensions from that? I think will be key to perfect stock alignment...

tbfleming, 2017-08-14:
Matrix is more general than yaw,pitch,roll. I don't think there's a reliable way to go back.

jorgerobles, 2017-08-14:
😥

jorgerobles, 2017-08-15:
@tbfleming there is hope https://forum.openframeworks.cc/t/artoolkitplus-get-translation-rotation-from-matrix/4335/10

jorgerobles, 2017-08-15:
Do you know anything about http://www.babylonjs.com/ ??

tbfleming, 2017-08-15:
Nope

jorgerobles, 2017-08-15:
Seems there's a way to get the transform matrix to euler, but it's definitely out of my reach :|

mrcg, 2017-09-23:
Really nice feature. Would it be possible to have different pattern which triggers type of marker.
E.g.
* One pattern which triggers homing upper/left corner without rotation.
* Another pattern which triggers homing upper/left corner with rotation enabled.
* Having two patterns where upper/left (a) and upper/right (b) are marked and LW calculates the rotation from this two pattern.

Maybe also some other pattern which can be used to set homing trigger for center/left/right etc.
WDYT?

jorgerobles, 2017-09-23:
Yes that's planned ;)