LaserWeb/LaserWeb4 #355: could not open a svg file

lollotek, 2017-06-26:
Hi,
I think I found a bug, when I try to open this file i get an error.
I'm secure it was working using laserweb3

Branch es6-dev:
```
index.js:sourcemap:60266 Uncaught (in promise) TypeError: this.node.getCTM is not a function
    at Element.elproto.transform (index.js:sourcemap:60266)
    at addChildren (index.js:sourcemap:42410)
    at loadSvg (index.js:sourcemap:42494)
    at documentsLoad (index.js:sourcemap:42609)
    at Object.reducer [as default] (index.js:sourcemap:41860)
    at hot (index.js:sourcemap:16550)
    at index.js:sourcemap:41456
    at dispatch (index.js:sourcemap:38819)
    at index.js:sourcemap:41387
    at index.js:sourcemap:41542
```

[testsvg.zip](https://github.com/LaserWeb/LaserWeb4/files/1103505/testsvg.zip)

cojarbi, 2017-06-26:
It's does not open for me either unless I open it in inkscape and do  Object to Path

lollotek, 2017-06-26:
Ok, FYI using LaserWeb3/master it loads without inkscape editing.
Thanks for the support.
[image]

jorgerobles, 2017-06-27:
Seems some crap from Snap library. https://github.com/svgdotjs/svg.js/issues/385 :(

tbfleming, 2017-06-27:
This file is defective. Not even Chrome, which is pretty permissive, can display it. It starts with:

```
<svg width="400mm" height="300mm" viewBox="0 0 400 300">
```

It should be:

```
<svg xmlns="http://www.w3.org/2000/svg" width="400mm" height="300mm" viewBox="0 0 400 300">
```

Snap builds on Chrome's SVG handling; it breaks when Chrome breaks.

tbfleming, 2017-06-27:
@jorgerobles to test for corruption like this, drag the file into an empty Chrome tab.

jorgerobles, 2017-06-27:
Yes, I overlooked that. :/

lollotek, 2017-06-27:
I added xmlns attrib to my generator, and import works.
thanks

closed 2017-06-27 by lollotek