LaserWeb/LaserWeb4 #356: Work area / set zero odd behavior

openhardwarecoza, 2017-06-27:
1)  Machine has no endstops
2) Jog to where I want zero.
3) hit set zero
4) white are is now off way somewhere else and i got to squint to see what I am doing around 0,0,0 which is now in the gray part?    (; see why didnt ever expose users to coordinate systems, now set zero is less of a set zero more of a "fuck up my view"

openhardwarecoza, 2017-06-27:
Personally, i would want to opt-out of the gray area too...

openhardwarecoza, 2017-06-27:
(; i take it you only tested with Grbl?

cprezzi, 2017-06-27:
No, testet with smoothie and grbl-LPC too. I found a problem with zeroing on grbl-LPC, as G10... doesn't seem to work. Had to use G92 instead.
Did you find a different behaviour?

openhardwarecoza, 2017-06-27:
Looks like it.  Set zero sets the DRO to zero, the "white box" is just offset to where it was before I set zero. 

Any specific Smoothie settings?  This machine has pretty much stock smoothie config - steps per mm and acceleration modified. No changes around endstops/homing (none connected so didnt bother)

cprezzi, 2017-06-27:
We could implement a setting "Ignore Work Offset" or "Disable Machine Space" so the work offset would not be read from the firmware, but this wouln't solve the problem that the firmwares internal work offset could increase over time and could overflow the vars without warning.

openhardwarecoza, 2017-06-27:
Yeah makes sense.  G10 P2 probably only sets the G54 offset, and machine offset stays untouched

cprezzi, 2017-06-27:
This is like it should work: The white space shows the machine space and stays where it is. When you set zero, the axis origin moves to the location you have set (as setZero is setting the work offset).

If the firmware is initializing with machine coordinates 0,0,0 then you should manually move the head to lower left before starting it.

openhardwarecoza, 2017-06-27:
Hmmm, shouldnt White be the "current coordinate system" instead - ie what
you set as zero?  (The grid should be too) - and then we find another way
to show Machine Position.  Work Pos would be the more important one to see
, after setting zero you surely want your work area to be clean and white?

cprezzi, 2017-06-27:
We (Jorche, Todd and me) decided that the white space should represent the machine and the axis represent the workspace. This way, the user can see if his job maches the available machine space or if he has to shift the work offset.

cprezzi, 2017-06-27:
The problem with the old "workspace only" way is, that you could not realize that your job will go out of bounds.

openhardwarecoza, 2017-06-27:
Theory vs practice.  Jog, set zero, now you got to setup CAM against a dark
background :/

cprezzi, 2017-06-27:
My practice is: Homing, Jog to bottom left of workpiece, set zero, see on the screen if it fits the machine and run the job. ;)

If homing is not available: Switch machine off, Jog to bottom left of machine on screen, set zero, switch machine on, jog to bottom left of workpiece...

openhardwarecoza, 2017-06-27:
Again, theory vs practise. Cant turn off machine between churning out
parts.  Just jog over to unused part of stock, set zero, press play

cprezzi, 2017-06-27:
The "procedure" with switching off machine an jogging is not needed between two jobs, it only needed when you moved the head manually or the firmware is changing the offset.

openhardwarecoza, 2017-06-27:
Nevermind. i see i'll have to keep living in the dark. 

Not like you guys ever considered milling with 0,0,0 in the center of the stock, or jogging into negative wpos.  i'll fork and fix for myself

cprezzi, 2017-06-27:
Milling with the origin in the center of the stock was one reason why we had to implement work offset, and is also a reason why it doesn't make sense to highlight only the positive working space, but the machine space instead. If we have a real problem, we need to solve it, but I'm not sure this is the case.

@jorgerobles @tbfleming I think you should check this thread. I'm not sure if I understand something wrong. Do we have a major problem?

openhardwarecoza, 2017-06-27:
Leave it lying closed... if more users complain, we'll see - for now i'll
go install endstops. easier than arguing.

openhardwarecoza, 2017-06-27:
still think an optout for those without endstops would help. but oh well.
remember to document: "Burn out your drivers by turning machine off, moving
axes by hand, then turning it on / resetting controller..." - yes back emf
from motors kill drivers, ask @raykholo (;

Anyway - i'm not going to spend 10 mins twisting leadscrew axes knobs if i
turned machine off with it near axes max on X  and Y...   Nor do i want to
find something to poke the Smoothieboards reset button thats hidden next to
the endstop pins and cant be reached by hand... so installing endstops is
easier

openhardwarecoza, 2017-06-27:
final say:  I realise its a FIRMWARE limitation - we cant reset mpos.  My
objection is in forcing MPOS on users who may not know/need/have
working/care for MPOS

cprezzi, 2017-06-27:
Shure, moving head manually is not a genuine tip ;) but it's also no sooo risky. We do it permanently on the diode laser.

A setting like "homing disabled" would be nice, but we need to find out how we could do that, without risking the firmware offset overflow. I need to to investigate the different firmware behaviours. Probably I can find a "solution" to reset the offset.

PS: Using MPOS instead of WPOS would lead to not being able to set zero at all.

openhardwarecoza, 2017-06-27:
Yeah definitely dont use mpos.

tbfleming, 2017-06-27:
I'll add an option which does this:
* Whole display will be white
* Black machine bounds markers will disappear
* Nothing on the display will shift with Set Zero

I'm thinking of defaulting to this mode since it doesn't require the user set up the machine bounds (size and lower-left position settings). Machine vendors will handle the setup for their machines. End users with home switches can enable it if they understand how to configure it. Grid bounds will still go both positive and negative.

openhardwarecoza, 2017-06-27:
Happy with that.  Neg grid is fine, i use centered stock all the time (easier than corners when milling on existing pieces)

tbfleming, 2017-06-27:
Now if I could only stop users who don't understand MPos or WPos from making bizarre requests such as being able to enter a negative machine height...

openhardwarecoza, 2017-06-27:
Lmfao! Yeah i feel you!

raykholo, 2017-06-27:
You mean the request we get every Tuesday? 

Ray

openhardwarecoza, 2017-06-27:
Yip that one. (;

tbfleming, 2017-06-27:
> * Nothing on the display will shift with Set Zero

Maybe I'll keep the shifting. Might be less confusing than the purple position marker jumping locations.

openhardwarecoza, 2017-06-27:
Grid shiting to zero works fine. Its just the gray "border" that made working hard

cprezzi, 2017-06-27:
Wrong button ;)

tbfleming, 2017-06-27:
New setting: Show Machine

openhardwarecoza, 2017-06-27:
Thanks Todd. Closing for now. I trust that'll do

closed 2017-06-27 by openhardwarecoza