LaserWeb/LaserWeb4 #358: Declutter Raster CAM - filters duplicated, like new way better

openhardwarecoza, 2017-06-27:
[image]

Filters is kinda duplicated between CAM tab and the popup anyway... Maybe just keep the Popup one?  So much better with the preview, the CAM tab one kinda useless in comparison

I also feel these buttons doesnt belong on the POSITION modal:

[image]

I feel like they belong on the CAM tab?

jorgerobles, 2017-06-27:
Well they are not duplicated but seems to. I'll explain.
- Using filters on Operation makes nothing to the image itself, but on the result of the gcode operation.
- Using filters on Trace/Modify, alter the original image (if desired) and are part of tracing result. No way back except undo.

For example, you could want to keep the original image and tune the result of the gcode operation.

jorgerobles, 2017-06-27:
About the "non positional buttons" on the floating window, I see with other perspective: All of the options on the floating controls are modifiers to the document selected. I found this a very convenient place to put that tools, If not, some bar should be shown when documents are selected, or being permanently shown but disabled when not selected.

openhardwarecoza, 2017-06-27:
Re duplication: 

So, we dont have a preview of "raster" filters yet?  (If not, can we please get a modal too? its really cool) 

and b) then if I understand correctly the button launches a modal that should only be used for Trace? not for filters?  Still then feels like one shouldnt be there or the "filters" should be removed from the button text.  I bet it wont just confuse me, seeing the exact same dialog as in the cam tab

openhardwarecoza, 2017-06-27:
Re floating window:  i hear you.  Still feels like an odd place to launch a trace

jorgerobles, 2017-06-27:
Theres a raster effects preview where the gcode toggles. That makes preview of filters applied to operation

openhardwarecoza, 2017-06-27:
[image]
Ok found that... but i hope seeing it in thread like this, you start to see my point?  This preview is even kinda weirder? Perhaps if it pops up automatically when you click into one of the filter fields? (no way for us to just use the 3d viewer as the preview for the filters?

openhardwarecoza, 2017-06-27:
Goes along with the main point I am trying to make in #359 - our UI has gone a little off the rails, incoherent, various ways of presenting things, not seamless.  Great Backend though...

jorgerobles, 2017-06-27:
I agree that those bunch features seem unrelated or poor planned, but indeed are. :) We can always improve obviously

openhardwarecoza, 2017-06-27:
(: there we go. Thats why opened the issue, so we can improve the whole ui
(: - its gotten so complicated even Team OpenBuilds doesnt want to use it
anymore lol

openhardwarecoza, 2017-06-27:
Closing #358 since #359 covers the larger overhead problem here

closed 2017-06-27 by openhardwarecoza

domenic-d, 2017-06-27:
Just some input:
We have always struggled with the lack of vertical space when working with many operations. Constant scrolling, maximising, minimising makes the process somewhat frustrating.

Some suggestions for discussion:
1: Group common attributes, such as Filter settings into a collapsable group. The user can expand this if they need to make adjustments, otherwise all they see is 'Filters'.
This can be potentially be applied to other attributes as well and a 'standard' set of groups will be initially all that users will see.

2: Create a pop-out bubble for attributes within parameter groups. This will allow the vertical menu to remain a manageable size.