LaserWeb/LaserWeb4 #359: Hidden Features

openhardwarecoza, 2017-06-27:
Theres been a LOT of new features the last couple weeks, but so much of it is missing "intuitive" or "detail" from them:

Examples:
1) Right Clicks - sure we can come up with something better?  Not intuitive
[image]

2) Typing,  wait for result, type again...  hmmm, I wonder, sliders? - feel like in this use case it may work way better 
[image] 

3) Use Gamepad
- i think it could do with a modal popup that a) shows you detail of which button does what or b) a mapper to allow you to allocate axes/step sizes/something to buttons
[image]

4) Use Numpad:  Used to be to enable https://github.com/kabachello/jQuery.NumPad on selected text inputs.   I think i recall from a past issue it's now to enable use of the Numpad on aan actual keyboard for jog?   Same as 3 then, a popup to show what does what could do good?   
[image]

openhardwarecoza, 2017-06-27:
(; more importantly - tabs are cramped - the use of Modals works well, like you did for tracing.  More of that can only be a good thing

jorgerobles, 2017-06-27:
> 5) Raster Filter Preview - perhaps use the 3D Viewer itself? Didnt even know about the popup before i asked Jorge - its not seemless, its not in the users face like it should be

Was planned that way, but doing so implies to duplicate the dataUrl of every document. Being raster could be a lot of data.

openhardwarecoza, 2017-06-27:
makes sense yes. darn.  Still, sleep on how to make it more clear.  The current raster preview i dont like.  I do like the Trace modal.   I think that has potential for being the fix for (5) too

jorgerobles, 2017-06-27:
> Right Clicks - sure we can come up with something better? Not intuitive
Hmm it has the context menu icon on hover :|

openhardwarecoza, 2017-06-27:
Right click (and taking it further, us Pipo users)  isnt really a popular thing in webapps. And latest binary build doesnt show context cursor on hover. Even if it did, i feel like people wouldnt try rich clicking on a static label. doesnt make sense.

jorgerobles, 2017-06-27:
Ok will see if there's a way to trigger on press for that component

openhardwarecoza, 2017-06-27:
(: I know i sound pedantic - but UI is hard to get "right"

jorgerobles, 2017-06-27:
> 2. Typing, wait for result, type again... hmmm, I wonder, sliders? - feel like in this use case it may work way better

On my way

openhardwarecoza, 2017-06-27:
Super Fast (;

- but no need to rush sliders, consider that part of rethink for 6 (;

jorgerobles, 2017-06-27:
True. But I was completely agree for 6, so I made it :D

openhardwarecoza, 2017-06-27:
Lol cool. Glad you liked the sliders idea (;

jorgerobles, 2017-06-27:
> 4

[image]

jorgerobles, 2017-06-27:
Well 6 is more complicated. I made for number 2 :)

jorgerobles, 2017-06-27:
@openhardwarecoza Cannot use 5 as you say (preview on canvas).
Example:
You have 2 operations with the same document added to them. In one, you engrave the image as usual. In the second operation you want to maximize contrast to make darks darker. 
The image preview should be contextual to the Operation, not the document.

tbfleming, 2017-06-27:
Here's what happens:
* We design a UI around a planned feature set.
* A ton of feature requests come in which don't fit the plan. No requester sees the harm with adding what they want.
* UI becomes confusing and crowded.

Potential solutions:
* Strip out a bunch of functionality to simplify the UI
* Try to come up with a new UI plan which better handles all the new functionality. This is **extremely** hard. Not the coding part, the coming up with a UI concept part.

jorgerobles, 2017-06-27:
Completely agree. You can blame freely :smile:

tbfleming, 2017-06-27:
Today is the first time I found out what that dotted blue line is. I've been assuming it was a CSS glitch that I planned to fix when I rework the CSS.

cojarbi, 2017-06-27:
UI option: have you seen Rhino or Grasshopper? They use tabs at the top to group all the gazillion options and tools available.

jorgerobles, 2017-06-27:
LOL.Well in my discharge some people found it appealing.
@openhardwarecoza if you keep pressed over the dashed text about a second it already pops up the menu!

openhardwarecoza, 2017-06-27:
@tbfleming is spot on.   It happens, no hard feelings, but its at a point where it is a problem, i think we can all agree. Not right away necceserily, but we need to start thinking of a solution.  I do love all the power features, but on the other hand i feel like the "power" makes some if it confusing.  The art lies in presenting just enough to a user, while still showing them the option of all the power if they need it.  I totally agree thats not easy.  

In a way, i regret the sidebar idea we went with - it wasnt scalable enough. My bad, that was me in LW3 (;

jorgerobles, 2017-06-27:
> UI option: have you seen Rhino or Grasshopper? They use tabs at the top to group all the gazillion options and tools available.

One of the pieces of software where I feel deeply lost, the other one is Blender.

cojarbi, 2017-06-27:
I hate Blender UI also. Forget Rhino, what about Excel for example

openhardwarecoza, 2017-06-27:
(; kidding aside, my favourite CAM UI is LW.  So its not like theres something i want to say "we can draw inspiration from that"... very hard

openhardwarecoza, 2017-06-27:
I dont think we're that far of track... just a little cleanup here and there may just do...

cprezzi, 2017-06-27:
I love all the cool features we have, but it should no scare beginners away. 
Unfortunately I am not a big frontend guy, but could we implement some operation mode like beginner, advanded, pro (superman) and filter some features by the mode?

openhardwarecoza, 2017-06-27:
Exactly yes.  beginner should see "just enough" and advanced should be under a <kbd>advanced</kbd> <random div type here> where random div is either collapse, modal, replace beginner div with and advanced div, etc

jorgerobles, 2017-06-27:
> under a advanced where random div is either collapse, modal, replace

Surprisingly some begginers asks for some advanced features

openhardwarecoza, 2017-06-27:
Yeah, but then they can click <kbd>advanced</kbd> (;  - the keyword was "some" -   the other 90% got dazzled lol

tbfleming, 2017-06-27:
Here's the problem:
* New user starts in simple mode.
* New user asks in forum how to do X. e.g. convert bitmap to vector.
* Answer: switch to advanced mode.
* UI changes. New user is now confused. So are experienced users.

openhardwarecoza, 2017-06-27:
Something else i just though of:  Something  like what you just added inhttps://github.com/LaserWeb/LaserWeb4/issues/359#issuecomment-311398717

In fact thats the same problem - it just added clutter to the workspace.  While at the same time it failed to 100% convey function.  you and I know X is left/right so we'll know to use 4/6 for left right.  A newb doesnt.  

Case in point: (;  - my neighbour:
[image]

Every function basically needs some "help" - instead of adding little notes that clutter up, but dont tell the whole story, can we perhaps consider create a new standard component called "helpbox" or something.  It could be a tooltip, or a modal, that pops up and then has space for pics, text etc.  I was working on this just before you posted the fix (; (didnt finish)

[image]

Then we can unify the "where to find the info" into a standard style on every element (a <kbd>?</kbd> perhaps, that onclick, or onhover opens the helpbox?

openhardwarecoza, 2017-06-27:
@tbfleming (; advanced shouldnt change the UI that much that it leads to confusion... when properly laid out it just adds fields in a way that adds power without confusion.

cojarbi, 2017-06-27:
@openhardwarecoza what are all those cable coming from the ceiling? :)

openhardwarecoza, 2017-06-27:
@cojarbi the place i am renting already had a piece of wood nailed to the wall with cup hooks screwed in.  the various cables (usb, network, power, earphones) just hang from those hooks

Heres a bigger view 

[image]

Those hooks in the ceiling was from the landlord making "biltong"  - ie meat hung from the ceiling and dripped on the floor.  Had to clean all that out, repaint, etc before i could move my tools in... smelled horid

jorgerobles, 2017-06-27:
So you managed to sell him a ob router?

openhardwarecoza, 2017-06-27:
@jorgerobles not yet.   he's checking mine but eish, LW too hard, its making my selling difficult man (;

jorgerobles, 2017-06-27:
Well sounds like the landlord did not know about Jamón Serrano

openhardwarecoza, 2017-06-27:
Negative... neither have I.  Biltong is a very South African thing... everyone, always, has biltong in the house... Game meat (from Kudu, Springbok, Eland) is cheap around here (one of the perks of the place i moved to... check https://www.youtube.com/watch?v=PKLHCaQL3EQ - thats just 5 mins from my house) hehe

cojarbi, 2017-06-27:
"LaserWeb too Hard" 
@openhardwarecoza that's a Tyson punch

openhardwarecoza, 2017-06-27:
yeah it felt like Tyson hit me lol.   But no blame, he was a train engineer all his life, but retired 30 years ago.  Before the days of CNC. so i actually had to explain XYZ to him lol

jorgerobles, 2017-06-27:
@openhardwarecoza ->
https://www.culinary.co.za/food/spanish-foods/olm285602-andalucia-jamon-iberico-de-bellota-deboned-36-48-mths-per-kg

openhardwarecoza, 2017-06-27:
@jorgerobles thats half of what my rent for a 4 bedroom house, with pool, double garage and outside workshops cost... , for just 1Kg!   I got to figure out how to make it, and find a market to sell it to (;  - this looks profitable!

jorgerobles, 2017-06-27:
Spanish soccer team smuggled some back on 2010. Here in spain is about
20-30 usd :)

tbfleming, 2017-06-27:
Here are some of the unplanned things which complicated the UI. I'm probably missing a lot because I really can't remember everything that was added:

* Heavy document modification: Raster to vector. I only planned for simple modifications: load, move, scale, delete. I didn't plan for rotate, but it fit in OK. I also didn't plan for shortcuts to optimize raster; they kind of fit, kind of don't.
* Preview the image adjustment settings on the raster ops. I'm very uneasy about this being present since the Laser Preview should have been enough.
* Presets for machines and materials
* A display which reliably shows machine bounds when Set Zero is used, assuming home switches are present. It was the commercial vendors which pushed hard for this, but under an unworkable request which clashed badly with rotary. The current display handles both, but at the expense of machines without home switches.
* Keyboard shortcuts: both machine control and document editing.

I pretty much ignored all of the following during initial design since it didn't involve the Files tab (was Cam tab), even though some of it had discussion early on.
* Macros
* Live Jogging
* Joystick
* Webcam overlay window

jorgerobles, 2017-06-27:
Balloon 
[image]

openhardwarecoza, 2017-06-28:
Baloon looks good - less intrusive than the inline help, dont you all agree (; ?