LaserWeb/LaserWeb4 #360: Make position window draggable

openhardwarecoza, 2017-06-27:
Often times I like to ctrl+click entities in the viewer.  But if you forget to work from bottom left, very often the position window obscures the entitity you need to click next. Clicking away, clears the selection. So you end up having to do a weird select something to move it away, deselect it, then select then thing you wanted to select.  Perhaps a small title bar at top of that window, that can be used as a drag handle, drag it out of your way, then, when you click the next entity, it defaults back to lower left?

cojarbi, 2017-06-27:
@openhardwarecoza did you see #348

openhardwarecoza, 2017-06-27:
Only now that you pointed me to it...

So then i guess my question is "until you drag the floating from the crosses icon. " mean what? I cant figure out where to grip it to drag it?

cojarbi, 2017-06-27:
Not sure if i understand the question but i just drag it to the top left corner of the grid and its stays there, even after restart.

openhardwarecoza, 2017-06-27:
What on the position window do you click on to start dragging it:

I tried gripping it by the gear icon, the gray background, the buttons, nothing reacts to trying to drag it

cojarbi, 2017-06-27:
The cross arrows at the top left

openhardwarecoza, 2017-06-27:
Can you screengrab for me?  I dont have it?  (reinstalled win last week, downloaded latest release this morning, thats why i find so many things now - havent updated in weeks

jorgerobles, 2017-06-27:
http://recordit.co/f29gRt1itG

openhardwarecoza, 2017-06-27:
negative... i dont have it. not in newest binary yet?

openhardwarecoza, 2017-06-27:
Will recheck after a new build. Can't download the whole build env right now - internet getting throttled because of "10 day rolling usage fair usage policy" - grr

closed 2017-06-27 by openhardwarecoza