LaserWeb/LaserWeb4 #363: Constant offset messages ... still

KirAsh4, 2017-06-27:
I opened an issue about this not too long ago and despite the best efforts from @cprezzi this has yet to be resolved, and since I can't seem to be able to re-open an issue, I decided to post about it again.

I don't know if it's now considered normal for these offset messages to continuously be scrolling in the console window or not but at least for me, they are 1. useless, and 2. in the way. If I don't remember to look at the window to capture the length of time a job took, there's no way to find it again. Once I hit the 'home all' button, the console gets filled with more offset messages. This only happens as the machine is actually running and performing an operation, whether it's simply homing, or cutting/engraving. It also seems to get more and more sluggish as the job progresses. Usually on something like this, I have to completely exit LW and relaunch before moving on to the next one.

I posted a video here about it. Note the console window: https://www.youtube.com/watch?v=gtkBoVsChwM

jorgerobles, 2017-06-27:
@KirAsh4 You could reopen the old one, just remove `is:open `and search for the issue

[image]

jorgerobles, 2017-06-27:
@KirAsh4 please send the file and tell me your board, firm, and general setup. If Smoothie, attach your config.txt, if GRBL please run $$ and send the dump. Also dump the workspace to check any conflicting LW setup.

I've followed this issue since opened the last time and could not replicate behaviour in any board I've got (UNO, SmoothieWare and GRBL-LPC)
Offset message should happen only once, When calling Set Zero, or homing. 

Thanks!

KirAsh4, 2017-06-27:
It was issue #324 when I first posted about it. Both you and Claudio helped. And I remember the note about "it should only happen once" but the stream of messages proves otherwise. It happens with *any* file, whether I'm cutting or engraving, or even just when homing the unit. Doesn't seem to be anything related to the file. The board is a Cohesion3D Mini running "Firmware smoothie edge-8e4a64c" (this I get from connecting the machine in Comms)

The attached zip file has the SVG I was cutting when I took that video, as well as config.txt and the workspace JSON file.

[Issue363.zip](https://github.com/LaserWeb/LaserWeb4/files/1106631/Issue363.zip)

KirAsh4, 2017-06-27:
I had also created a logfile at that time, and noticed it too is filled with a constant stream of Offset messages and Claudio said that's normal:
https://github.com/LaserWeb/LaserWeb4/files/1065447/logfile.txt

jorgerobles, 2017-06-28:
@KirAsh4 First check. Do you have the firmware-cnc bin ? http://cncpro.co/index.php/30-documentation/initial-configuration/firmware/smoothieware

KirAsh4, 2017-06-28:
Yes I do, taken from here
https://github.com/Smoothieware/Smoothieware/tree/edge/FirmwareBin

KirAsh4, 2017-06-28:
By the way, I've recently seen at least 1 screen capture from another user that also show the same streaming messages on their system, so I know it's not just me.

jorgerobles, 2017-06-28:
I'm on it :) just don't know why my machines don't reflect that behaviour

jorgerobles, 2017-06-28:
Found and Fixed.

jorgerobles, 2017-06-28:
@cprezzi some values are received from client as string and should be Numbers (floats or ints). That causes problems when NaN and "NaN" are involved :P

closed 2017-06-28 by jorgerobles

KirAsh4, 2017-06-28:
Awesome. I'll wait for the next binary release to test. Gracias Jorge!

cprezzi, 2017-06-28:
@jorgerobles I think we could just deactivate the work offset messages completely. They was just needed for testing.

cprezzi, 2017-07-11:
I have reworked the whole work offset part in lw.comm-server which should have solved this issue.
(https://github.com/LaserWeb/lw.comm-server/commit/d0808a708700830ab5eb794e32082ac645ef926f will be included in next release)

KirAsh4, 2017-07-11:
Awesome. I noticed they were less in the last release, but still popped up every so often, for example during homing, I would get a half dozen or so messages popping up. Don't remember if it continued during a job, but I want to say probably not. I haven't used my cutter since I last updated LW.

cprezzi, 2017-07-11:
The workOffset message should only popup once after the offset has changed, i.e. after `set zero` or `homing`.

KirAsh4, 2017-07-11:
So in my experience, this is what happens:
- Turn machine on
- In LaserWeb, connect to the machine - FIRST OFFSET MSG POPS UP
- In LaserWeb, click 'Home All' - SEVERAL OFFSET MSGS POP UP

This last step has LESS messages now than in the previous release. And after I start a job, I will see one or two more pop up, then it stops. In the previous release, the messages continued to pop up while the job was running. So at least that has been resolved somewhat.

commit aac84a31 (2017-06-28): Fix #363