LaserWeb/LaserWeb4 #364: Gamepad enhancement

openhardwarecoza, 2017-06-27:
Continuation from #359 :  GamePad

I can jog XY just fine, but two suggestions:

1: Step size modifier: Seems to always jog 1mm:  Perhaps holding down one of the buttons on the right side of gamepad, could act as a modifier for 0.1?  Especially on milling, Z jogs are almost always 0.1.  Maybe another button for x10?  Or a button that cycles through 0.1, 1, 10, 1, 0.1 and displays it onscreen? 

2: Cant seem to jog Z yet? 

and 

3:  Possible UI to remap own controller. i have a Buffulo NES pad, so my right hand side buttons are in a straight grid. PS style controller they are in a + grid. So probably no one size fits all.

openhardwarecoza, 2017-06-27:
Gamepad mapping demo: http://charliejwalter.net/gamepad/demos/main/  (SVG of gamepad with inputs overlayed? )

openhardwarecoza, 2017-06-27:
Second note, just tried my USB snes joystick, but forgot to unplug the NES. Realised we may need a select box like we have for Webcam, to pick Gamepad

jorgerobles, 2017-10-21:
Freezing feature for now. No reports of usage.

closed 2017-10-21 by jorgerobles