LaserWeb/LaserWeb4 #367: SVG embedded image displaced vertically

allstar87, 2017-06-30:
Attempting to create an object which is a raster image centered on a vector cut.  Created in Inkscape v 0.91. Created the following SVG:
[basicsaturn5.zip](https://github.com/LaserWeb/LaserWeb4/files/1116111/basicsaturn5.zip)

The blue rectangle and the image of Saturn both have their lower left corners on the origin.

When imported into Inkscape, the embedded image is shifted vertically 3mm, as seen here:
[laserweb_inkscape_screenshot.zip](https://github.com/LaserWeb/LaserWeb4/files/1116135/laserweb_inkscape_screenshot.zip)

[Laserweb-Workspace.json.txt](https://github.com/LaserWeb/LaserWeb4/files/1116114/Laserweb-Workspace.json.txt)

Only Laserweb shifts the image like this, opening the SVG in other software displays the image correctly.

tbfleming, 2017-06-30:
The Laserweb-Workspace.json.txt file above is empty.

Here's what happens when I load basicsaturn5.svg into LW:

[image]

allstar87, 2017-06-30:
That's what I'm trying to achieve, and failing.  I'll attempt to reupload the workspace json.

tbfleming, 2017-06-30:
Very odd. Go ahead.

tbfleming, 2017-06-30:
Sounds like you might be getting file system corruption.

allstar87, 2017-06-30:
[Laserweb-Workspace.json.zip](https://github.com/LaserWeb/LaserWeb4/files/1116200/Laserweb-Workspace.json.zip)

tbfleming, 2017-06-30:
I can tell from the .json that you're on an older LW version. I've done major changes to SVG import since then.

allstar87, 2017-06-30:
From the About section:  Frontend 4.0.731, backend 4.0.68.  I'll go fetch the most recent appimage I can find.

allstar87, 2017-06-30:
That got it.

jorgerobles, 2017-07-05:
Still an issue? if not please close

closed 2017-07-06 by allstar87