LaserWeb/LaserWeb4 #368: Virus warning

yztink, 2017-07-02:
Sorry, I'm sure this isn't the right place to submit this, but I don't know where would be better.

I tried the latest two windows 64bit binaries, and both were caught by my virus scanner. I doubt there's really a virus, but it unnerves me....has something changed that would make it do that or is something compromised? In the past I ran version, 4.0.68-729, with no issue (and I just ran it again from my computer and it works fine).

I use Panda antivirus on Windows 10.

jorgerobles, 2017-07-02:
Please attach virus info

jorgerobles, 2017-07-02:
Avast throw no virus.

jorgerobles, 2017-07-02:
No virus found on win10 panda free av.

yztink, 2017-07-02:
Hmmm, weird. So I took a closer look and it quarantined the exe files becausee they are "suspicious files" not viruses. Still weird that it detects those and not the older build.

yztink, 2017-07-02:
1409749515	129505	6	 Sun Jul 02 00:56:27 2017	 Action result	 ON ACCESS EXECUTION	 A:\Downloads\LaserWeb.Setup.4.0.734-96-x64.exe	 -	 Blocked	 01-07-2017 07:58	  Suspicious	 4AB69C21DF2D9C53A346AFF2FB97A6F8	 0x03078000	50	3749687	 -	8	 -	 -	 -	 -	20

yztink, 2017-07-02:
From the log file...Not sure what it means though lol

timestamp,job Id,profile,date,job type,task type,file (path),pid,result

1409744640, 129504, 1, Sun Jul 02 00:56:22 2017, Analysis result, ON ACCESS EXECUTION, A:\Downloads\LaserWeb.Setup.4.0.734-96-x64.exe, 12056, 01-07-2017 07:58, Suspicious, 4AB69C21DF2D9C53A346AFF2FB97A6F8, 0x03078000, 50, 3749687, 0, 8, -, 0, 0, -, 20

jorgerobles, 2017-07-02:
Thank you but AV is not very helpful. As far as I've tested on a couple of AVs too and no warning, going to close the issue. Thanks for the info.

closed 2017-07-02 by jorgerobles

nathanielstenzel, 2019-11-22:
Is this by any chance related to index.js. Not like it is a recent copy, but I was doing a ClamAV scan and it said something about dist/index.js having an Html.Exploit.CVE_2016_3271-2. The timestamp in the folder was February 27, 2017.

cprezzi, 2019-11-22:
This exploit seems to be related to IE11, but we build the exe with Electron, which embeds a chromium browser. So there should be no risk of this exploit.