LaserWeb/LaserWeb4 #369: OSX - permission denied on 'logfile.txt'

gerwinbrunner, 2017-07-03:
I'm running LaserWeb 4.0.735 on OSX.
When I enable the log level and set it to 3 then I get the following error message:
[image]

It seems that this is an issue with rights on OSX.

Here is how I do this:
```
export LOG_LEVEL=3
open LaserWeb.app
```

gerwinbrunner, 2017-07-03:
I think the problem is this call here:
https://github.com/LaserWeb/lw.comm-server/blob/41f0844fe40e69f20c6373ab904c0e006e2362dc/server.js#L2797

The path should somehow be passed in (or set for this library) by using a platform specific path like `app.getPath('userData')`
The electron api I'm talking about is here:
https://github.com/electron/electron/blob/master/docs/api/app.md#appgetpathname

jorgerobles, 2017-07-04:
Yes, I've have the same issue.
Could you lend us a hand to fix it? Thanks!

jorgerobles, 2017-07-06:
Moved to lw.comm-server

jorgerobles, 2017-07-06:
BTW. the patch works ok

`logFile = fs.createWriteStream(path.join(electronApp.getPath('userData'),"logfile.log"));`

closed 2017-07-06 by jorgerobles