LaserWeb/LaserWeb4 #372: 3D Milling :P (Sorry Todd)

openhardwarecoza, 2017-07-06:
Related to https://plus.google.com/+ArielYahni/posts/iCjFq7t62aK

I remember Todds work from months ago (offsetMesh https://tbfleming.github.io/OffsetMesh/ and also some work he did in CP) 

Theres also this project : https://github.com/damianmoore/part-mill by @damianmoore (if we need hands on deck, we can ask (;) 

And of course @nraynaud (who i borrowed Drag Knife maths from for lw3) has https://github.com/nraynaud/webgcode which can also do a little 3d it seems? 

(; now it's an open issue. I say lets start a fundraiser and buy the dev team some OpenBuilds MiniMills (; (+-$500 per person x 3 devs) @cojarbi (we had to use Vectric for https://plus.google.com/+Openbuildspartstore/posts/Epqc6YD7Pgs)

nraynaud, 2017-07-06:
not to be bragging but webgcode has the most complete 3D milling system of the bunch. In particular because there are various tool shapes supported.

openhardwarecoza, 2017-07-06:
@raynaud any chance we can bribe you into coding for us? I am sure we can
organise you a mill or a laser?

nraynaud, 2017-07-06:
now, we're talking :)
To be fair, I'd love me some gear, but 
1) I'm moving continent next month and will be without a permanent place for a while, 
2) I'm an independent contractor, and if I receive "donations" in exchange for work, the tax man might want to have a quick chat with me.

On the serious side: for now I'm committed to other projects, and I'm afraid that telling that I will help you and then not walking the walk would be worst that saying no now.

But I understand that the 3D milling code from webgcode is hard to follow. 

I left some testing code: https://github.com/nraynaud/webgcode/blob/gh-pages/various_tests/test_3D.html ( http://nraynaud.github.io/webgcode/various_tests/test_3D.html )
I think it could help by limiting the scope of what to copy/paste in webgcode (open the example and look at the list of source files in chrome debugger).

Nico

nraynaud, 2017-07-06:
http://nraynaud.github.io/webgcode/various_tests/test_3D_conservative_rendering.html here the demonstration of the conservative rendering, it show, that the tool (the little squares) will never violate the net shape of the part (the triangles), and the red vertices show how the conservatism is computed.

openhardwarecoza, 2017-07-06:
> http://nraynaud.github.io/webgcode/various_tests/test_
> 3D_conservative_rendering.html here the demonstration of the conservative

Cool man! I appreciate it! Will wait for @tbfleming to look things over
then someday (:

openhardwarecoza, 2017-07-06:
@raynaud when you get settled in, feel free to jump back in and ask for
Push rights anytime you want!

openhardwarecoza, 2017-07-07:
@tbfleming anything useful in the above that could make your life easier?

tbfleming, 2017-07-07:
I don't know. It takes time to evaluate existing code. I have a strong suspicion that in the 3D cam space, "use X", where X is something you don't like mentioned, will be the best option without years of work.

cojarbi, 2017-07-07:
Who is this X that we don't like being mentioned ( sound like Harry Potter movie or Beetlejuice )

openhardwarecoza, 2017-07-07:
(; i assume webasm? (;

Dont mind me, i'm so out of the project my opinion doesn't matter. Dont let
my unwillingness to learn new tech, get in the way of this project (;. I
say Go for it!

tbfleming, 2017-07-07:
No, I already started on moving toward wasm. Too many performance complaints to stick with pure JS for polygon ops.

X = Fusion 360

cojarbi, 2017-07-07:
@openhardwarecoza your opinion does matter!

openhardwarecoza, 2017-07-07:
Ahh ok. Yeah i dont mind it that much. But as a disgruntled Eagle user who
just had to fork over the Premium licence fee... I dont love Autodesk too
much

openhardwarecoza, 2017-07-09:
Well, closing for now (:

closed 2017-07-09 by openhardwarecoza