LaserWeb/LaserWeb4 #373: Conversational Lathe

tbfleming, 2017-07-06:
I've been eyeing doing this for a while, but there's a hardware roadblock:

* Low-cost lathes which are rigid enough (e.g. my Sherline and bench-top Chinese lathes) have backlash which is too large for CNC
* Lathes which are rigid enough and have low-enough backlash are out of the price range of typical LW users and have conversational baked into their control system
  * https://www.tormach.com/product_15l_slantpro.html
  * https://smithy.com/automatecnc/smallcncgangtoollathe/fh30b
* A long-shot idea which failed because of rigidity: use a Sherline headstock with C-Beam: https://plus.google.com/101442607030198502072/posts/Veb93ckULW1

openhardwarecoza, 2017-07-06:
What about a Router Lathe with CBeam (; the forces on a router lathe is a lot less than chisel types?

openhardwarecoza, 2017-07-06:
RE your CBeam flex issue:  Since then they have added http://openbuildspartstore.com/c-beam-gantry-plate-xlarge/ - the large polycarbonate wheels and being outside instead of inside the CBeam, makes these massively stronger.  If you want to try the experiment again, i think i can organise you a set of the plates and wheels (you still have the rest?)

tbfleming, 2017-07-06:
I used a pair of XLarge plates with the wheels that come with the axis kits. Is there a stackup which will allow the larger wheels to do this without increasing the distance between the two plates?

kram242, 2017-07-06:
Hi @tbfleming really cool project you have in the works. I have been wanting to try this one myself so its really good to see you have already been working on one and we can all learn together. First I would like to say thank you for all you do in the community and to let you know that if you reach out to us when your working up a project,  OpenBuilds would be more then happy to help support it so others can learn. I also would like to help out and see if we can work together to get it working. I have looked though all the pics and videos I can see but its hard to see how the plates are set up. If you could send me a few shots of the plates without the shield that would be helpful. Thanks in advance and I am really hopeful that this could work!
OpenBuilds conversation(PM) - http://www.openbuilds.com/members/mark-carew.1/#info
OpenBuilds Email - OpenBuilds@OpenBuilds.com

openhardwarecoza, 2017-07-06:
@tbfleming looks like you have the stock config, mini v wheels inside the cbeam?

tbfleming, 2017-07-06:
@openhardwarecoza yes. I'm taking photos now for @kram242

openhardwarecoza, 2017-07-06:
(: cool man. Yip xtreme vwheels (regular sized not mini) running in the
outside rails of the cbeam (like we did on the OpenBuilds Minimill) will be
a LOT stronger

tbfleming, 2017-07-06:
Posting here so everyone can see how I set up the axis.

[image]
[image]
[image]
[image]
[image]

openhardwarecoza, 2017-07-07:
Yip, can definitely be done better:

1: replace the included cbeam plates, with the XL plates (instead of
stacking 4 plates)

2: use the regular sized wheels (if you dont have, let us know so we can
get some to you

3: i would try bolting the second cbeam to the first xl plate using T nuts
and M5x10s (if you dont have let us know)

4: i would then put the tool holder on the carriage of the second XL plate.

I'll mock it up in Sketchup for you later today. Just woke up. Got a little
filming to finish for the morning, will tackle the drawing after lunch (:

<https://user-images.githubusercontent.com/418464/27930587-31a1d768-6265-11e7-8a71-ff435b7e7baf.jpg>
[image: b]
<https://user-images.githubusercontent.com/418464/27930598-3a8a4112-6265-11e7-844c-9622a696804f.jpg>
[image: c]
<https://user-images.githubusercontent.com/418464/27930603-3e9e861e-6265-11e7-9e52-7f67713caadd.JPG>
[image: d]
<https://user-images.githubusercontent.com/418464/27930610-42856176-6265-11e7-8fda-31b2fdb15c86.JPG>
[image: e]
<https://user-images.githubusercontent.com/418464/27930615-4654fbea-6265-11e7-9972-2cc60b972d0c.JPG>

tbfleming, 2017-07-07:
4: I want to mount the tool holder on the T-Slot side: a) so it can adjust and be replaced easily and b) so I can mount multiple gang-style

tbfleming, 2017-07-07:
Probably worth trying. If it works we can design a new plate to move the T Slot side back on top.

2: Don't have

3: I have over 40 T nuts and over 40 socket-head (not low profile) M5x10s left

openhardwarecoza, 2017-07-07:
Here's how I would build it (Sketchup attached inside zip) - based off the MiniMill XY table since I know its strong

That means you'd need 
8 x extreme vwheel kits
8 x corner connectors
1 x 8020 V-Slot (as a tool holder on top of the last plate) - 125mm offcut? or 250mm piece
1 x pack of 10mm low profile screws (to mount the C-Beams to the XL Plates) 
1 x pack of 8mm low profile screws ( to mount reinforcing corners to side of cbeam and 8020)

[Todd Lathe.zip](https://github.com/LaserWeb/LaserWeb4/files/1131099/Todd.Lathe.zip)

openhardwarecoza, 2017-07-07:
[image]

openhardwarecoza, 2017-07-07:
PS: I removed the C Beam Spacers, and Shield from the build that you had on your first one., to accomodate the heigh increase of the 8020.  Removing the extra plates and the spacers from the stack, i think this might be almost a less-tall setup that your original. 

Worried about chips on the leadscrews?  Paper bellows (http://openbuilds.com/projectresources/bellows-maker.234/) or a piece of kitchen sponge (http://openbuilds.com/builds/the-makersl-customized-ob-minimill.3647/) (;

tbfleming, 2017-07-07:
I'll try that with 250mm to leave room for multiple tools.

openhardwarecoza, 2017-07-07:
Gotcha, passing it on to Mark (;

openhardwarecoza, 2017-07-07:
Might need to space it up a little to clear the C-Beam Ends on the ends then...

tbfleming, 2017-07-07:
Just checked. It will clear the C-Beam ends, but not the stepper. I'll have to shim it about 1mm.

tbfleming, 2017-07-07:
The sponge won't protect the slides and wheels :(. I may have to do this again: https://youtu.be/K1Ka0ULu0as

kram242, 2017-07-07:
Thanks for the pics Todd, Peter is spot on in the MiniMill set up above this should give you the strength you need for the lathe build. You could also try the 'sandwich' approach like we did on the XL build.
The system is really flexible in the way it can built so there is a lot of freedom in the configurations.
Here are a couple more examples that may help.
[image]
[C-Beam connection concepts.zip](https://github.com/LaserWeb/LaserWeb4/files/1131207/C-Beam.connection.concepts.zip)

tbfleming, 2017-07-07:
The left stackup in @kram242 's post looks like the one with the least height. I think I'll try it first.

kram242, 2017-07-07:
Ok cool I will get some parts together for you to test out. I will also get the parts for Peters set up as well. I think its a good test as the ones I posted above are cantilevered style where the whole actuator moves instead of the plate. This could leave areas of deflection and was the main reason that we decided to go with the style Peter is showing above for the MiniMill. Its worth testing them both to see what works.

cprezzi, 2017-07-07:
Probably you can add a pvc roll mechanism to cover the bottom spindle from dust. :)

tbfleming, 2017-07-07:
@kram242 Thanks!

@cprezzi What is a pvc roll mechanism?

jorgerobles, 2017-07-07:
Bellow?

openhardwarecoza, 2017-07-07:
Something like https://youtu.be/P-aamGvgbd8 ?

tbfleming, 2017-07-08:
If this experiment works then the next step will be to look at spindle options. Right now I'm using the Sherline headstock with motor ($336 new). This doesn't support:

* Automatic speed control. There's a way to hack a 10V analog input into the motor's controller, but this runs it blind.
* Reverse. There's a way to hack this into the motor's controller, but it tends to cause pieces of the motor brushes to break free and fly into where they don't belong, frying the motor.
* Position and speed feedback. LinuxCNC people mount a sensor on spindles, but it only gives 1 pulse per spindle rotation. GRBL doesn't support this. I don't know if Smoothieware does.

I'd like to thread (barely works with above sensor using LinuxCNC) and maybe rigid tap (needs better spindle position control + reversing). To do this with GRBL or Smoothie it looks like I'll need a spindle drive with step/dir inputs.

Idea:
* Headstock:
  * Sherline headstock without motor ($108 new), or 
  * LittleMachineShop's Mini Lathe Headstock Assembly without motor ($120 new)
    * Bonus: chuck won't accidentally screw off running in reverse
* Servo motor + drive that accepts step+dir inputs

A Sureservo 400W Low Inertia System looks like it might have a good enough torque curve ($1049 not including power supply or back-current protection).

Anyone know of lower-cost servos?

tbfleming, 2017-07-08:
It looks like Teknic ClearPath has much better continuous torque/$, easier install, but a lower peak RPM (4000, sherline=6000). 
* CPM-SDHP-3432D-ELN = $655
* IPC-5 power supply = $248

tbfleming, 2017-07-08:
I suspect the there'll be two major build variants:
* Less expensive: servoless headstock
* More expensive: servo headstock
   * Threading
   * Rigid tapping
   * Speed varies with cutter position to produce better surface finish

I'll fund the servo purchase.

tbfleming, 2017-07-08:
Tooling: Sherline's tall T-nuts loosely fit V-Slot. It works, but I suspect that frequent changes will eat into the slot. The lathe should be able to produce a replacement tall T-nut that fits better.

tbfleming, 2017-07-08:
[image]

tbfleming, 2017-07-11:
@kram242 How do you remove the bearings from wheels? I need to reuse some for the extreme wheels. I also forgot it insert the shim when I pressed together one of the extreme kits, so I need to get that one apart also.

tbfleming, 2017-07-11:
Got it; took many tries with a screwdriver on the mistake. Then I noticed the bearings for the existing wheels are a different size. I'll try 4 wheels on each plate instead of 6.

tbfleming, 2017-07-12:
I tested @kram242 's left config with 4 wheels on each plate. Much better. The visible wobble is gone when cutting, but it hits a nasty resonance, even with the eccentrics at their tightest positions. I tried a variety of speeds and feeds trying to get it out of the resonance range, but was unsuccessful finding a reliable recipe. Some recipes managed to make some nice cuts in between their nasty ones, something my original couldn't do.

Next step: @openhardwarecoza 's config.

tbfleming, 2017-07-12:
https://plus.google.com/101442607030198502072/posts/RBLFvKg7jGo

tbfleming, 2017-07-12:
Progress. I switched to @openhardwarecoza 's stackup, but left out the offcut for now. I'm trading off tooling flexibility for rigidity. It still has a bad resonance, but it shifted. The entire resonance used to be on the X slide; now about half of it is on the spindle support instead. This means it's time to stack the spindle directly on the Z slide.

tbfleming, 2017-07-13:
Sherline riser block mod: a mix of metric and imperial:

* 5.5mm holes, 60mm apart
* 3/8" endmill, .4in deep
* m5-30 screws

The height isn't a perfect match; I'll use the rocker toolholder to compensate. Quick-change toolposts can also compensate.
[image]
[image]
[image]
[image]
[image]
[image]

tbfleming, 2017-07-13:
Dang. Still no good.

tbfleming, 2017-07-15:
@jeepshop and I were trying various things today and came up with this: https://plus.google.com/101442607030198502072/posts/PV3BiYvrxaL

It's okish with ~12mm stock, but ~17mm is still unusable. If we could increase the rigidity more, I believe we'll be there. @kram242 : have you thought of producing delrin insert which slides inside the extrusion channel?

[image]

It would match the plate's length. Screws would pull this up toward the plate, preloading the wheels in the direction they're weakest.

tbfleming, 2017-07-15:
Woohoo! We used a file to smooth some t-nuts, gave them some oil, and used them like the delrin insert idea. Getting tension right is tough since they want to lock things down.

https://plus.google.com/photos/photo/101442607030198502072/6443096668177628226

tbfleming, 2017-07-23:
Here's it running with LW-generated gcode:

https://youtu.be/cR0OX3ZF2jo

tbfleming, 2017-07-23:
I just found this today; I should have looked earlier: Taig added ballscrew models last year. $1625 including steppers. Add a smoothie or grbl controller and tooling and it should do what mine does above, but with heavier cuts and a good finish.