LaserWeb/LaserWeb4 #374: Bugfix release

tbfleming, 2017-07-07:
@jorgerobles could you publish a new exe release please? I upgraded snapsvg to fix a bug some users were hitting.

jorgerobles, 2017-07-08:
https://github.com/LaserWeb/LaserWeb4-Binaries/releases/tag/v4.0.736-114

jorgerobles, 2017-07-08:
BTW - I no longer compile to all platforms from osx since toolchain broke a couple of releases ago (so have to switch desktop/laptop/os in the meantime). No worries, but if anyone is interesting building a platform, please do.

I strongly recommend the use of git tags to mark new versions. Just `npm version patch` Is very useful to difference versions and compose the release summary (I'm looking at you @cprezzi! 😄 )

jorgerobles, 2017-07-08:
@tbfleming  some snapsvg dependency broke binary build

[image]

tbfleming, 2017-07-08:
Shoot.

tbfleming, 2017-07-08:
I just built the installer on Windows 10 from fresh clones. It appears to be OK. How can I open the dev tools on the exe to check the log? shift+ctrl+i doesn't work like it does in Chrome.

tbfleming, 2017-07-08:
I'm on Node v6.10.3.

jorgerobles, 2017-07-08:
On about, open dev tools

tbfleming, 2017-07-08:
Yep. Connection refused. I don't get the window not defined error.

I also tried your installer: LaserWeb.Setup.4.0.736-114-x64.exe . Same thing: connection refused, but no window not defined error.

Both yours and mine load an SVG file OK which caused issues with the old Snap version.

tbfleming, 2017-07-08:
@cprezzi Any idea what would cause Connection Refused?

tbfleming, 2017-07-08:
Ooooh. Generate gcode broke.

tbfleming, 2017-07-08:
Fixed the snap/webworker issue in branch fix_374. The ajv stuff doesn't build correctly on my system. I have no idea what's causing Connection Refused.

jorgerobles, 2017-07-09:
Thanks @tbfleming was late night for me 😴

tbfleming, 2017-07-09:
Heh. I had the wrong USB cable plugged in. I no longer get Connection Refused!

tbfleming, 2017-07-09:
I just published v4.0.988-114 win-64 version.

tbfleming, 2017-07-09:
Are there any 32-bit users left in the world?

jorgerobles, 2017-07-09:
LOL. :)

cojarbi, 2017-07-09:
I have a win XP 32 bit

tbfleming, 2017-07-09:
Eeek! I wouldn't want an XP machine anywhere near a network after what's been going on the past few months.

jorgerobles, 2017-07-09:
@tbfleming should branch fix_374 merged on dev-es6?

tbfleming, 2017-07-09:
@jorgerobles yes.

jorgerobles, 2017-07-09:
Panda Free AV reports W32/Exploit.gen :cry:

tbfleming, 2017-07-09:
My guess: someone created a Trojan using Electron and the imbeciles at Panda were stupid when creating the pattern.

jorgerobles, 2017-07-09:
Avast reports its ok. Yes, panda av is not my favorite.

tbfleming, 2017-07-09:
It's that kind of stupidity on AV pattern makers which creates situations like this: https://www.theregister.co.uk/2017/04/25/webroot_windows_wipeout/

mayhem2408, 2017-07-09:
@tbfleming I have an old PC running an old linuxcnc machine and it runs XP 32-bit. My other PC in my workshop runs my 3d Printer, laser engraver, laserweb4, cnc.js all on Windows 10 64-bit.

jorgerobles, 2017-07-09:
I've compiled win32 and osx too.

domenic-d, 2017-07-10:
There are unfortunately many users we are aware of who are running win32 systems.

closed 2017-07-10 by jorgerobles