LaserWeb/LaserWeb4 #377: High CPU & RAM consumption

msurguy, 2017-07-14:
Hello! 

I am using the latest version of LaserWeb and am getting very high CPU / RAM consumption on both, Electron and hosted versions of the program. I have a 2.5Ghz quad core i5 and 16GB of RAM, with no other programs running besides LaserWeb and seems like there is a memory leak or some event listeners that are maxing out the CPU cycles.

Fun fact, this is really helpful if you want to discharge your laptop's battery in less than 10 minutes!

jorgerobles, 2017-07-14:
@tbfleming 
---
For reference:
- https://addyosmani.com/blog/taming-the-unicorn-easing-javascript-memory-profiling-in-devtools/

tbfleming, 2017-07-14:
@jorgerobles yep, I often use those tools. Problem is I usually develop on a high-TDP desktop which doesn't have those symptoms.

tbfleming, 2017-07-14:
e.g. I have the .exe version up now. It consumes ~0.5% while idle.

tbfleming, 2017-07-14:
Turning on the Display Cache drops it to ~0.1%.

jorgerobles, 2017-07-14:
@msurguy Could we have your computer profile and workspace.json? Any data you can provide us to debug? Do you have the camera installed?

tbfleming, 2017-07-14:
@msurguy things which may affect this:
* CPU model
* GPU model, if you have a discreet one
* OS, version, and 32-bit or 64-bit

Most reports I've seen so far with this problem are with old hardware or old OS. A 2.5 GHz CPU produced within the past couple years behaves very differently than a 2.5 GHz CPU produced 7 or 8 years ago.

msurguy, 2017-07-14:
@jorgerobles attached is a screenshot of my Mac profile: 

[image]

Could be that this issue is only on Chrome Mac. 

The issue appears even when I don't do anything other than open the app and let it sit idle.

jorgerobles, 2017-07-14:
My desktop is similar :/

[image]

jorgerobles, 2017-07-15:
try chrome://gpu/ to detect problems

[image]

[image]

msurguy, 2017-07-20:
Thanks @jorgerobles! Tried to check for GPU problems and none were found. 

What I did discover is that this problem seems to appear only sometimes, and only when I have an external monitor plugged into my MacBook. When the monitor is unplugged, the CPU load is very minimal!

jorgerobles, 2017-07-20:
Glad to hear. Unfortunately due the randomness of the issue cannot offer any further solution. Closi g by now.

closed 2017-07-20 by jorgerobles

tbfleming, 2017-07-22:
I suspect this might be happening: adding an external monitor increases the amount of video ram dedicated to running the display. This leaves less video ram for the GPU to use for handling LW's requests. The CPU is forced to take up the slack.