LaserWeb/LaserWeb4 #379: LaserWeb crashes on setting minimum grid size to 0

wdehaas, 2017-07-15:
I'm using LaserWeb4 Mac OSX version.

I tried to set the minimum safe grid offset size to 0 and then the program crashed. I think I did this under: 

Settings: Machine

Now when I try to start the program again I get a blank screen. Tried to reinstall the program but somehow some settings remain I think. Does someone know where I can find the setting files in OSX?

Thanks in advance for the reply.

greetings,

Waldemar

jorgerobles, 2017-07-15:
~/Library/Application\ Support/lw.comm-server/Local\ Storage

jorgerobles, 2017-07-15:
Before deleting, please attach .localstorage files to this issue to debug

wdehaas, 2017-07-16:
Hi,

Thanks for your reply. As far I can see I cannot find the /lw.comm-server directory under Application Support. Are you sure this is the right location for it?

Thanks for your reply.

jorgerobles, 2017-07-16:
On mine is :|

closed 2017-10-21 by jorgerobles

kreaziku, 2018-07-21:
this problem is happen to me too :( blank

commit 607a3869 (2017-10-21): Fix #379