LaserWeb/LaserWeb4 #380: NOT_A_BUG! But the cat is curious about released files ...

KirAsh4, 2017-07-15:
Are the zip and tarball on the Mac/Win binaries download link meant to be empty? All they have in them is the README.md file. If they are meant to be empty, may I suggest at least adding a note about where the source can be fetched from?

I know how to do this, but for those who don't ... although there's an argument there too whether they ought to be mucking with the source if they can't figure out where it is. :)  (in which case I'd suggest removing them altogether.)

tbfleming, 2017-07-15:
GitHub puts those there automatically.

KirAsh4, 2017-07-15:
Odd that they would put (nearly) empty source zip/tarballs up. Some things aren't meant to be figured out ...

KirAsh4, 2017-07-15:
The naming on the Windows binaries are also messed up. They're missing the .989 in the name like the other binaries do. Gremlins!

jorgerobles, 2017-07-16:
Not gremlins. Just copy pasted wrong editing release

closed 2017-07-23 by jorgerobles