LaserWeb/LaserWeb4 #381: Compiling problem - buffertools

domenic-d, 2017-07-17:
We have started not being able to compile the latest version on Win with the following error being prevalent:
ENOENT: no such file or directory, open 'C:\Users\DarklyBase\lwo\LaserWeb4\node_modules\buffertools\package.json'

Tried fresh clones with same results.
Has anything changed with the compiling requirements that we need to know about?

domenic-d, 2017-07-17:
@tbfleming @openhardwarecoza 
Can you please kick off a binary build until I get this issue resolved for local compiling?

tbfleming, 2017-07-17:
Starting a build now for win64.

tbfleming, 2017-07-17:
I built from fresh pulls and didn't run into that problem.

https://github.com/LaserWeb/LaserWeb4/releases/download/v4.0.988/LaserWeb.Setup.test.vendor.presets.w64.exe

domenic-d, 2017-07-17:
Any ideas on why we could be getting this?
We are running Win 10 x64.
Any new dependencies we may be missing?

tbfleming, 2017-07-17:
No idea. I rarely use npm's -g option to reduce the chances that I have a dependency installed that's not in package.json. I'm also on Win 10 x64. Node 6.10.3.

tbfleming, 2017-07-17:
Did you try from fresh clones?

domenic-d, 2017-07-17:
Yes, but I will try them again now.
Thx for the build. I'll let you know if I discover anything.

domenic-d, 2017-07-17:
Ok, looks like we are not getting that issue now. Rebuilt from fresh clones today without an issue.
The only difference is that I ran through the process manually rather than via the build script we normally use.
Thx for the help and I will close this off.

closed 2017-07-17 by domenic-d