LaserWeb/LaserWeb4 #385: Operation processing & Filtering

domenic-d, 2017-07-19:
@tbfleming 
If a very complex vector file is opened in Laserweb, would there be any advantage to using filters to process sections of the file in multiple operations, rather than in one operation?

In other words, is there any computational advantage in creating multiple operations but limiting the region for each?

[image]

cprezzi, 2017-07-19:
I'm not absolutely sure, but it could be faster to have multiple operations because multi threading works better then (depenting on your CPU and thread setting).

jorgerobles, 2017-07-19:
Yes. I think I mentioned in other issue, current threading system works on
multiple operations.

tbfleming, 2017-07-19:
Probably need to try it and see what happens.

closed 2017-10-02 by domenic-d