LaserWeb/LaserWeb4 #386: Accumulated Job Time only saved to store when closing app

cprezzi, 2017-07-25:
If the app is closed after each job, the job time is correctly added to settings.jogAccumulatedJobTime, but if a job is run multiple times (or several jobs are run), it always adds the job duration to the accumulatedJobTime from app start.

@jorgerobles There must be something wrong with the dispatch(setSettingsAttrs..) call in line https://github.com/LaserWeb/LaserWeb4/blob/dev-es6/src/components/com.js#L404. The value of settings.jogAccumulatedJobTime is not updated after the dispatch line. Do I have to refresh settings somehow after the dispatch(setSettingsAttrs()) ?

tbfleming, 2017-07-25:
The objects in the store are read-only. https://github.com/LaserWeb/LaserWeb4/blob/dev-es6/src/components/com.js#L60 captures the settings object, which never changes. https://github.com/LaserWeb/LaserWeb4/blob/dev-es6/src/components/com.js#L404 causes the store to create a new settings object. Instead of capturing the settings object, always reference this.props.settings.

tbfleming, 2017-07-25:
Since function callbacks clobber ```this```, you'll need to use the old ```self``` hack or switch to arrow functions for the callbacks.

cprezzi, 2017-07-27:
The self hack (var that = this) was not working and changing everything to arrow functions would be too much work, so I solved it with the help of a global var ;)

closed 2017-07-27 by cprezzi

domenic-d, 2017-07-27:
@cprezzi That's great news. thank you.
Can you tell us how this time can be queried?
How can it be reset?

Thanks again.

cprezzi, 2017-07-28:
@domenic-d The value (comAccumulatedJobTime=seconds) is saved in the settings.json which can be exported. To change the value you just have to change it in the file and reimport it.

Don't add this param to the default machine profiles! Otherwise the clients data will be overwritten with a LW4 update.

domenic-d, 2017-07-31:
Thx @cprezzi