LaserWeb/LaserWeb4 #387: Check Size button doesnt honor XY Jog Speed

openhardwarecoza, 2017-07-26:
### Follow this steps to open an issue ###

Sometimes one may want to run fast, but sometimes a second time a little slower to make sure everythings lined up.  Changing XY jog speed seems to have no effect on Check Size anymore?  (in LW3 I used to...https://github.com/LaserWeb/deprecated-LaserWeb3/blob/8cc62c5278faa9dbb8794ce792adaeebdba6b89d/public/js/jog.js#L22)

openhardwarecoza, 2017-07-31:
@cprezzi (;

cprezzi, 2017-08-02:
The *check size* button does respect the actual XY jog speed (tested on my grbl-LPC system).
This is the used code:
```
let feedrate = $('#jogfeedxy').val() * 60;
let moves = `
    G90\n
    G0 X` + xMin + ` Y` + yMin + ` F` + feedrate + `\n
    G1 F` + feedrate + ` S` + power + `\n
    G1 X` + xMax + ` Y` + yMin + `\n
    G1 X` + xMax + ` Y` + yMax + `\n
    G1 X` + xMin + ` Y` + yMax + `\n
    G1 X` + xMin + ` Y` + yMin + `\n
    G90\n
`;
```

What firmware are you using?

openhardwarecoza, 2017-08-02:
Smoothieware

cprezzi, 2017-08-02:
Ah, I sse what the problem is. The feed calculation doesn't respect the configured feed units, so the calc is wrong when using mm/min ;)

openhardwarecoza, 2017-08-02:
Yay (: so i am not (entirely) crazy yet

cprezzi, 2017-08-02:
Nope

cprezzi, 2017-08-02:
Ok, done with commit https://github.com/LaserWeb/LaserWeb4/commit/b75b4ffda8e8a0d688202f1894dfa256928e5d97

closed 2017-09-28 by cprezzi