LaserWeb/LaserWeb4 #388: Milling cam:  If smaller pocket inside bigger pocket, and bigger pocket is already cut: Can we make smaller pocket Z start at bottom of bigger pocket if depth > than big pocket?

openhardwarecoza, 2017-07-27:
Milling cam:  If smaller pocket inside bigger pocket, and bigger pocket is already cut: Can we make smaller pocket Z start at bottom of bigger pocket if depth > than big pocket?

This becomes a problem, when i am milling alu where generally the Z depth per pass is small.   Now you waste a lot of time milling "air" until you get to the bottom of the first pocket. 

[image]

openhardwarecoza, 2017-07-27:
... Even if its a manual entry Z start height

tbfleming, 2017-07-28:
I thought I had that, but it's not there. It won't take much to add it.

openhardwarecoza, 2017-07-28:
Yay (:

tbfleming, 2017-07-30:
Mill operations:
* Removed
  * Final Cut Depth
  * ZClearance
* Added
  * Rapid Z (the plane where it does XY rapids)
  * Start Z (first cutting pass is at Start Z - Pass Depth)
  * End Z

New capabilities:
* More flexible Z: top of material no longer forced to Z=0
* You can avoid air cutting by setting Start Z to previous operation's End Z

Be careful:
* It rapids Z down to Start Z, just like it used to rapid Z down to 0
* So you'll crash the cutter if your Start Z is below any remaining material

openhardwarecoza, 2017-07-31:
Thanks Todd.  Will test tomorrow!

openhardwarecoza, 2017-08-06:
Working great!

closed 2017-08-06 by openhardwarecoza