LaserWeb/LaserWeb4 #390: "Unknown Error Code 87"  While connected to C3D via LW4 will disconnect requiring reconnection.

Bama-Breeze, 2017-08-04:
While connected to C#D board via USB with LW4 connectivity with disconnect and an error code:  "Unknown Error Code 87".   Repetitive issue.  I can reconnect and in approximately 15-20 minutes of idle time will loose connection again.  No difficult reconnecting, but in a few minutes loose connection.  No projects being executed during connectivity issue.  There are NO power savers or other known issues or applications that would institute the issue.  I still have access via Windows Explorer to open the board's SD card.  Running Windows 10 and lasts version of LW4.  Have validate no issues with Com port.  Have switched between USB 3.0 and 2.0 with no difference.  Have changed USB cable with no difference.

Thanking you in advance!

Photograph to follow.
[image]

domenic-d, 2017-08-05:
@Bama-Breeze Can you please try to emulate this problem without Laserweb running or connected?
We have experienced similar issues with smoothieboards even when laserweb has not been running.

Can you please let us know the brand of laptop you are using? We have found this offers most often on Dell Inspiron machines.

Bama-Breeze, 2017-08-05:
Domenic:

My computer is a Inspiron Desktop 3668 purchased a month ago.  All if my desktop and laptop computers are Della.  All running Windows 10.  

Upon an effort to reproduce the issue I do not note replicated transient connectivity issues.  With Laserweb4 powered off and nothing more that the USB cable connected to the C3D and observed through Windows Explorer I did not observe instability and could see the SD card.  I tested the board with the transfer of a large 11mb pdf file from desktop to the board's SD card.  The 11mb transferred and was viewable without difficulty.

It may be a WASP, but I noted upon research on most recent Windows 10 updates that USB ports were exhibiting similar indiscriminate disconnects.  The fix was to go into the Device Manager and independently disable or deselect "Allow This Device To Deselect To Save Power" for each USB ports.  Simply turning off all power management simply through the Power Saving scheme was not adequate. 

It's beyond my skill level, but as I understand the issue they are describing the USB ports are going into a suspended state in power savings if it appeared a USB port's power source was not necessary in an idle state.  This again maybe little more than a WASP; however, one could surmise it as applicable since each time LW4 connectivity was in an idle state after a period of testing functionality of jog control of Y, X, Z, and A rotary.  The only device was the webcam, mouse, and keyboard.  As specified I observed no connectivity issues.  I monitored the webcam via an independent application.  Again I could open and execute a file in the on board AS care, i.e., I could open and view the 11mb pdf file I had pasted.

I will disable all power saving options on each USB ports later today in Windows 10 Device Manager.

If I have failed to follow your instructions please advise.  My doctorate is not in computer or data science.  While I possess a working knowledge I am most definitely not of the intellectual expertise as you and your colleagues.  So be gentle. 😊

I await your further thoughts, instruction, or chastising if applicable.

Bama-Breeze, 2017-08-05:
>   C3D via LW4 will disconnect requiring reconnection. (#390)

Domenic:

My computer is a Inspiron Desktop 3668 purchased a month ago.  All if my desktop and laptop computers are Della.  All running Windows 10.  

Upon an effort to reproduce the issue I do not note replicated transient connectivity issues.  With Laserweb4 powered off and nothing more that the USB cable connected to the C3D and observed through Windows Explorer I did not observe instability and could see the SD card.  I tested the board with the transfer of a large 11mb pdf file from desktop to the board's SD card.  The 11mb transferred and was viewable without difficulty.

It may be a WASP, but I noted upon research on most recent Windows 10 updates that USB ports were exhibiting similar indiscriminate disconnects.  The fix was to go into the Device Manager and independently disable or deselect "Allow This Device To Deselect To Save Power" for each USB ports.  Simply turning off all power management simply through the Power Saving scheme was not adequate. 

It's beyond my skill level, but as I understand the issue they are describing the USB ports are going into a suspended state in power savings if it appeared a USB port's power source was not necessary in an idle state.  This again maybe little more than a WASP; however, one could surmise it as applicable since each time LW4 connectivity was in an idle state after a period of testing functionality of jog control of Y, X, Z, and A rotary.  The only device was the webcam, mouse, and keyboard.  As specified I observed no connectivity issues.  I monitored the webcam via an independent application.  Again I could open and execute a file in the on board AS care, i.e., I could open and view the 11mb pdf file I had pasted.

I will disable all power saving options on each USB ports later today in Windows 10 Device Manager.

If I have failed to follow your instructions please advise.  My doctorate is not in computer or data science.  While I possess a working knowledge I am most definitely not of the intellectual expertise as you and your colleagues.  So be gentle. 😊

I await your further thoughts, instruction, or chastising if applicable.

domenic-d, 2017-08-05:
@Bama-Breeze we have also experimented with the power saving features etc, with no improvement.

This issue has been previously discussed with the lw.comms developer and laserweb lead, without any clear determination as to whether LW could be causing this. We since moved on and started looking elsewhere for a reason. The one link we have found is that it is related to Dell Inspiron computers.

I would recommend you raise this issue on the C3D forum, as I suspect it is not related to Laserweb.

Please share any findings you have and perhaps we can use them to find an answer to this.

Bama-Breeze, 2017-08-06:
Has this been an issue with every LaserWeb build?  Concerning since Dell computers are one of the leading consumer computer manufacturers by principal and other consumer brands.  

I will share with C3D.  I thank you, but somewhat disappointed in a reply indicative of an ongoing issue unknown to users.

tbfleming, 2017-08-06:
@Bama-Breeze it's only been within the last month or so that @domenic-d spotted the pattern. It doesn't appear to be all Dells, just some. It may not be just LW, it may be all apps which talk to Smoothieware when running on those Dells. We just don't know yet because we don't have enough data to go on and the lw.comms developer and the Smoothe developers don't have machines with that problem to debug with.

tbfleming, 2017-08-06:
A point of comparison:
* nVidia is a leading GPU maker
* Philips is a leading monitor maker
* It took 2 years before nVidia fixed driver bugs that caused my Philips 4K monitor to:
  * glitch every couple hours
  * randomly stop working at 60 Hz and fall back to 30 Hz, which strains my eyes
* Intel had similar bugs with this monitor

cprezzi, 2017-08-06:
Link to the post in the Cohesion3d G+ community (with my answers): https://plus.google.com/u/0/117720577851752736927/posts/jR3nBCbQCsb

This is what we know so far:
- Bad quality or long USB cables could cause it.
- Insufficent voltage on USB port could cause it (if board MCU is not powered from a separate source).
- USB 3.0 can cause it. Use USB 2.0 ports only.
- Sometimes just changing the USB port did help.
- Smoothieware can cause it, because of dual connection over the same USB port (CDC & MSD). Disable MSD (SD card interface) to resolv that or try replace Smoothieware with grbl-LPC.
- Mains power spikes can cause it (like light, air-condition or other equipement switching on/off)

openhardwarecoza, 2017-08-06:
The number part of the node-serialport is only reported by us. The error code is from the O/S:

87: https://social.msdn.microsoft.com/Forums/en-US/ad18e2a1-c25b-43e7-8b8d-019073db2c6f/why-setcommstate-error-87-?forum=vssmartdevicesnative

Bama-Breeze, 2017-08-08:
Gentleman if I may please accept my brief commentary.  As stated previously I have little more than "walking sense" in the technology you discuss herein.  My education resides elsewhere.  But, I so wish I could comprehend your discussion thread.  An academician in heakthcare for 40+ years I find your process extrodinarily  fascinating.  A process we would have called a "root cause analysis".  I'll step back into the shadows while continuing to enjoy your process.  Though I know not the subject matter I do appreciate the intellect.  Thank you for permitting my audience. -  My best regards to all!

Bama-Breeze, 2017-08-08:
Forgive my typos.

LdyMox, 2017-08-28:
I had this exact same problem with Windows 10 and a Dell  XPS 8910 with the Cohesion3D Mini.  When I switched over to grbl-lpc (https://cohesion3d.freshdesk.com/support/solutions/articles/5000740838-grbl-lpc-for-cohesion3d-mini) I haven't had the issue since, granted it's only been about 4 days but I did lots of projects over the weekend.   After you install grbl in LaserWeb for the machine settings you need add M4 to the start gcode, M5 to the end Gcode and Home to $H and PWM scale from 1 to 1000.

Bama-Breeze, 2017-10-03:
Thank you LdyMox.  How goes the experience with your switched over to grbl-lpc on your C3D?

closed 2017-10-03 by Bama-Breeze

LdyMox, 2017-10-03:
It has been 1000 times better since moving over to grbl-lpc from Smoothie on my C3D