LaserWeb/LaserWeb4 #391: [Question] Static Image below workspace

jorgerobles, 2017-08-05:
As a spinoff of Workspace video feed, an idea came from @domenic-d to include a ghost image under the workspace (in case video feed is too cpu intensive) 

The function is somewhat simple. Load an image and put below workspace, with an alpha setting.

Where do you think is the best place to it (In case there's a more convenient place that settings, as in files or so.)

tbfleming, 2017-08-05:
I don't understand. What good is a static image?

jorgerobles, 2017-08-05:
Well I see it useful as reference (some apps call it tracing image).

Also, related to camera feed, once the stock is in the desired position,
just a snapshot of that position could be useful for reference, without the
need of a cpu intensive webcam processing.

The Muse laser use a kind of snapshot to let the user put the artwork over
a certain position captured with a camera.

tbfleming, 2017-08-05:
Maybe a Freeze Camera button to reduce cpu?

jorgerobles, 2017-08-05:
Yes, but not limited to. As the work of image drawing on workspace is
indeed done, adding the capability to load an image to display is just a
couple of lines