LaserWeb/LaserWeb4 #392: Messaging

tbfleming, 2017-08-09:
I've seen several posts recently from frustrated people who expected to plug in a board, run the software, and everything to just work like an expensive integrated system. We're not that. We'll never be that. Vendors may use us as part of that, but users who install their own boards should know they have a lot of obscure and poorly-documented work ahead to get it running. Much more to get it running well.

How can we control expectations? This is the type of user that shouldn't upgrade their own machine: https://plus.google.com/107177313666688527432/posts/ixzQ186AWYU

tbfleming, 2017-08-09:
Another type of user that's wrong for us: the person expecting to run a business on their laser. They blow up at us when they discover we can't handle their clients' files because their clients didn't produce their files a suitable way.

jorgerobles, 2017-08-09:
Very difficult. Our software is not infalible nor any other component in the chain.
I read all the thread of this user, and is not one of my consideration, just by the tone he's using. I would tell him politely to look for an expert to solve his machine issues, as seems not a few, and probably he has some part of them.

Second type should be directed politely to the epilog web page. Merely an hour of support costs about a dozen of k40 :D

cojarbi, 2017-08-09:
Well I have responded twice to this user . I fully agree on what's stated above. But most importantly users must be mature enough to accept when thing are beyond their understanding. Sometines I do get myself in that kind of situation but I don't go ranting out there about it. Everything has a learning curve including replacing a board that is "drop-in"

Maybe we can write a welcome-disclaimer type of document that's explains this ?

jorgerobles, 2017-08-09:
I don't know if He or other type of user ever reads the disclaimer
even so, they will say they have the right to express their disagree with anything. Sadly, kind of these guys are so much time consuming to ever enjoy answering the community.

cojarbi, 2017-08-09:
Agreed Jorge. But not because of these users you'll deny the other group that thus read of the information that can guide them in the process. Now I'm not saying a disclaimer to remove responsibility to us (we are not responsable )

arthurwolf, 2017-08-09:
One thing it's good to remind users of whenever relevant, is they didn't
pay for the software ( or firmware for that matter ). They paid for the
hardware and if that's broken they deserve support from whomever sold it to
them ( not the community ), but the software is free, and made by
volunteers.
Open-Source problems can sometimes run into trouble with users feeling
"entitled" to things ( possibly because they are used to for things they
actually paid for ). The Smoothie project has a ( extremely perfectible )
explanation for users who get upset :
http://smoothieware.org/troubleshooting#i-m-very-upset-at-something-everyone-is-mean-and-nobody-listens-to-me
Maybe LW needs something like that ( written the LW way ) ? A disclaimer
when the program starts about responsibility and what kind of support to
expect is also possibly a good idea. Smoothie can't do that, but if it
could it would.

jorgerobles, 2017-08-09:
Agreed!

tbfleming, 2017-08-09:
I need friends to drink beer with. Well maybe not beer; I can't stand the taste!

cojarbi, 2017-08-09:
Digital beer maybe?

jorgerobles, 2017-08-09:
> I need friends to drink beer with. Well maybe not beer; I can't stand the taste!

Don't worry I will drink your beer for you! 🍺 
Have to work on that teleport machine again. Then there will be whatever drink you want to have :D

funinthefalls, 2017-08-09:
Want a bottle of 12 yr old Canadian Rye whiskey Todd?

cprezzi, 2017-08-09:
We need to arrange a LaserWeb Dev/Supporter meeting someday. I would love to take a drink (or two) with you guys!

cprezzi, 2017-08-09:
@openhardwarecoza Do you like to report about your new development you told me? This could realy help with config noobs...

openhardwarecoza, 2017-08-09:
Hmm. Not really (: (different methodology here, release after its ready till then secrets) - it ties into the original post though. Its a vendor solution to a vendor problem. Which can make its way back into lw (after all i need your help to get it into lw.comm-server) 

I'm on my phone now will post a sneaker pic in a few

openhardwarecoza, 2017-08-09:
You didnt get these from me if anyone asks....

[image]
[image]

Back to the original post - i can sense you guys are nearing the point i was a month or two ago.   Its not sustainable.  

My old attitude wasnt appreciated, but i would have no objection telling that kind of user to go fuck themselves.   The Cohesion3d is also a major source of this problem (big sell-up about how great and easy and plug in its going to be... yet then the fallout heads our way - another case in point here) 

Before i quit (which if i have to be honest, has been HEAVEN!  Not dealing with users who make me feel like shit all the time, has really helped me get my Fu' back and the pipeline is running with upcoming stuff again) i had the idea of maybe closing down support completely.  Ie you help yourself and thats it.  If you buy a significant portion of beer, i'll give you an hour of support via a support ticket system... Never went that way as our tireless support was part of our massive growth after all.   But people ruin it for us who carry the burden.  Instead of spending energy making LW better, we just burn out. Case in point since i left G+ and you guys took over the support load, look how the contiributions graph and "new features" started dipping!   Its a fact.  

Vendors:  As (now that i am an outside) an outsider i feel like "Pay up, or fuckoff.   If you make money you can afford your own support team, or pay us to help... otherwise your business plan is wrong"

openhardwarecoza, 2017-08-09:
Just a seperate view, particular user, again too old for this game, but there are valid points there, where even i complained about how LW4 became too complicated: 

His grid issue is Mpos related... If you jog jog jog, close lw, open again. - happens to me too. Resetting the controller to get mpos back to 0,0,0 fixes the grid. Or turn off mpos.    Thats why i always made a point of hiding mPos from users.  Those advanced features that make LW more "cnc machine" needs to stay hidden under an Advanced option, not on by Default.   Go back and look at lw1-3 and see what made this famous (; - simplicity

Documentation: We really could say "no support for the next month so we can use that energy and make good demo vids and good documentation" - even if its just to point to when someone then asks. Right now theres nothing to point someone to... I have that same problem recommending LW/CNCW to someone inside the OB ecosystem

tbfleming, 2017-08-09:
Grid 0,0 is WPos 0,0. The grid shifts when WPos shifts relative to MPos, but other than that, users don't have to deal with MPos unless they turn on Show Machine, which now defaults off. I recommend people without homing switches don't turn on Show Machine.

His issues go way beyond the grid; it's not homing correctly, throwing everything off. It also looks like it's losing steps (the wildly-changing sizes).

UI complications: the only way I know to simplify the UI is to strip away capabilities. Maybe we should create a simple app that does exactly 1 thing: raster 1 image. Not multiple. Not cutting. Maybe not even machine control; move the laser head by hand to where you want the lower-left corner.

Documentation: we need a volunteer for that.

openhardwarecoza, 2017-08-09:
Just saying since we brought that in, i also experienced grid weirdness.  

Agreed on his other issues though: That's Rays job to document and supply sane configs. 

Capabilities removed: I dont agree - but a lot of help towards newbz goes into wizards, tours, guesses.  We get the serial port pid/vid: If it matches a Smoothieboard, already autoconfig the stuff smoothie absolutely needs (pwm=1, etc) for the user. require just enough info on cam to get the job done.  Add a wizard to allow you to do some setup checking (jog and draw animation - compare, did yours jog the same way) etc.   A lot of those ideas come from work I am doing for OB - but not a line of code is written yet - just weeks of Thinking and counterarguing to decide how we'll do it.

openhardwarecoza, 2017-08-09:
```
function friendlyPort(i) {
    var likely = false;
    var img = 'usb.png';
    var note = '';
    var manufacturer = availports[i].manufacturer
    if (manufacturer == `(Standard port types)`) {
      img = 'serial.png'
      note = 'Motherboard Serial Port';
    } else if ( availports[i].productId && availports[i].vendorId ) {
        if (availports[i].productId == '6015' && availports[i].vendorId == '1D50') {
          // found Smoothieboard
          img = 'smoothieboard.png';
          note = 'Smoothieware USB Port';
          if (wizardBoard == 'Smoothieboard') {
            likely = true;
          } else {
            likely = false;
          }
        }
        if (availports[i].productId == '6001' && availports[i].vendorId == '0403') {
          // found FTDI FT232
          img = 'usb.png';
          note = 'FTDI USB to Serial';
        }
        if (availports[i].productId == '6015' && availports[i].vendorId == '0403') {
          // found FTDI FT230x
          img = 'usb.png';
          note = 'FTDI USD to Serial';
        }
        if (availports[i].productId == '606D' && availports[i].vendorId == '1D50') {
          // found TinyG G2
          img = 'usb.png';
          note = 'Tiny G2';
        }
        if (availports[i].productId == '003D' && availports[i].vendorId == '2341') {
          // found Arduino Due Prog Port
          img = 'due.png';
          note = 'Arduino Due Prog';
        }
        if (availports[i].productId == '0043' && availports[i].vendorId == '2341' || availports[i].productId == '0001' && availports[i].vendorId == '2341' || availports[i].productId == '0043' && availports[i].vendorId == '2A03') {
          // found Arduino Uno
          img = 'uno.png';
          note = 'Arduino Uno';
        }
        if (availports[i].productId == '2341' && availports[i].vendorId == '0042') {
          // found Arduino Mega
          img = 'mega.png';
          note = 'Arduino Mega';
        }
        if (availports[i].productId == '7523' && availports[i].vendorId == '1A86') {
          // found CH340
          img = 'uno.png';
          note = 'CH340 Arduino Fake';
        }
        if (availports[i].productId == 'EA60' && availports[i].vendorId == '10C4') {
          // found CP2102
          img = 'nodemcu.png';
          note = 'NodeMCU';
        }
    } else {
      img = "usb.png";
    }

    return {
      img: img,
      note: note,
      likely: likely
    };
}
```  

for example:  Extend that to do some automated guesswork and add a wizard on top.  

A first-run modal with info etc too

tbfleming, 2017-08-09:
Please try to find a reproducible way to make the grid act weird and file an issue.

jorgerobles, 2017-08-09:
Recently I've seen the work of  @sdesalas (DarklyLabs) about a setup wizard. It has to round some corners but I think it will be a good start. If you want to take a peek it's on its fork.

openhardwarecoza, 2017-08-09:
Will do! Since i stopped showing machine it hasnt happened yet.  Been freezing over here the last two weeks so tried staying OUT of the workshop! (; so my testing/cutting has been standing still

tbfleming, 2017-08-09:
I thought Darkly had a single config; load their profile and everything's set.

openhardwarecoza, 2017-08-09:
@jorgerobles (; any chance you have it on hand already for a screencap?  (Seperate issue though, maybe the UI one)

openhardwarecoza, 2017-08-09:
Another thing i am thinking off, any chance that user isnt on the latest, and thus show machine is still one (default) and therefore he is seeing grid issues? 

Pehaps something else worth doing first is reviving Auto Update so we can make sure users are on the latest all the time - and thus not opening issues / questions for things already resolved?  

https://github.com/electron/electron/blob/master/docs/api/auto-updater.md

jorgerobles, 2017-08-09:
http://recordit.co/8ZgPZgJ94T

As you can see it's meant to guide thru the connection/load/and basic job creation

openhardwarecoza, 2017-08-09:
Thats looks great! Love the comms tab with the flashing exclamation (; - immediately tells me i;m not connected.  What do we do?  Oh we hide it inside a tab where no one looks (;

jorgerobles, 2017-08-09:
Coming soon on the upcoming Lw release :P

closed 2017-10-30 by jorgerobles