LaserWeb/LaserWeb4 #393: Drawable Machine Control

tbfleming, 2017-08-12:
Moving to separate issue. Related to #378 and https://github.com/LaserWeb/lw.comm-server/issues/47

Current idea:
* End users may draw control UIs using Inkscape and post them online
  * Custom layouts
  * My be geared toward specific machines
  * My be anything from simple UIs targeted at beginners to full UIs supporting the entire set of gcode offsets for users with bigger needs
* https://github.com/LaserWeb/lw.controller will provide:
  * A non-UI component for managing comms
  * A connection toolbar
  * A component for displaying and hooking up SVGs
  * A stand-alone version with the above and a configurable SVG URL
*  https://github.com/LaserWeb/lw.comm-server will:
   * Provide more status and config info. e.g. the entire set of gcode offsets
   * Unify its messaging
* LW4 will:
  * Use lw.controller's non-UI component
  * Use lw.controller's connection toolbar
  * Use lw.controller's drawable UI in 2 locations:
    * Existing tab: compact display with configurable SVG URL
    * New tab: full-screen display with configurable SVG URL

tbfleming, 2017-08-12:
Jogging's now working with configurable distance and feed for each level.

[image]

jorgerobles, 2017-08-12:
Cool! Lovely work!

domenic-d, 2017-08-12:
@tbfleming How does this system work with the current LW interface?
Does it replace it, overlay on to it, sit beside it?
Do you have any plans to add in a workspace view element?

tbfleming, 2017-08-12:
@domenic-d both. It will be a standalone app, a small version will replace the existing control tab, and a large version on a new tab will cover the whole screen. No plans for a workspace view element since the small version will be to the left of the workspace.

domenic-d, 2017-08-13:
@tbfleming That sounds great.
What level of control will developers have? 
For instance, can they create their own UI to have:
     Ability to load/save workspaces?
     Ability to load gcode?

tbfleming, 2017-08-13:
It's only for machine control. It can load gcode and edit settings but won't touch the workspace, documents, or CAM.

cprezzi, 2017-08-13:
@tbfleming But we still need position update for the workspace to show the cursor position.

tbfleming, 2017-08-13:
@cprezzi com.js will update the store with the complete machine state. The workspace will monitor the store for the cursor position.

tbfleming, 2017-08-13:
DRO-like ability:

[image]

funinthefalls, 2017-08-13:
Looks good Todd