LaserWeb/LaserWeb4 #395: Raster Merge not respecting Intensity Settings

universam1, 2017-08-17:
`Laser Raster Merge` does not respect settings from `Laser Intensity` which results in invalid gcode in my setup.

I set `Laser Intensity =  M3 O`  which **works good** for `Laser Cut *` or `Laser Fill Path` giving following gcode:

```
; Pass 0 Path 0
G0 X30.67 Y0.37
M3 O255.00
G1 X29.79 Y0.37 F600
M3 O0
```

But `Laser Raster Merge` produces following code which is wrong:

```
G0 X29.08 Y0.09 S0.0000
G1 X29.13 S1.0000
X29.23 S11.0000
X29.33 S23.0000
```
In other word, it should not create Sxxx code but do 

1. INTENSITY SEPARATE LINE
1. M3 Oxxx commands

Thanks
[Laserweb-Workspace (6).json.txt](https://github.com/LaserWeb/LaserWeb4/files/1230300/Laserweb-Workspace.6.json.txt)

jorgerobles, 2017-08-17:
What controller are you using? Is this theorics issue or has indeed caused a real issue? (I've used Smoothie and GRBL and have not problem with inline Sxxx)

tbfleming, 2017-08-17:
Related to #334

universam1, 2017-08-17:
I'm using Marlin - I know it's not yet supported officially, but it doesn't prevent me from using it already **successfully** as long as I use `Laser Cut * or Laser Fill Path`

cprezzi, 2017-08-17:
@universam1 Cool. I did implement the marlinkimbra support in lw.comm-server several weeks ago but nobody tested it yet (as far as I know). Are there other issues you got with marlin?

universam1, 2017-08-17:
Oh wow! Sorry I'm pretty new with LW, so I'm using the github.io dist to create Goode then download and run via pronterface. 
So does it mean you got more support for Marlin developed. Could you give me a hint what to do in order to test your work, would love to!

cprezzi, 2017-08-18:
Ah ok, the way you worked did not use our server.

If you like to test streamig a job directly from LW4 to Marlin, you would just have to use our binary installer. It's already in there :)

The only expected thing is that the board responds with a welcome string like "mk_" when a M115 command is sent.

martindb, 2017-08-20:
I've made a MPCNC (https://www.vicious1.com) with a chinese laser led. I'm using standard ramps 1.4 with standard marlin (no kimbra). My laser is activated trought M106 Sxxx (PWM 0 to 255).

I can cut / fill with lw, using M106 S255 for gcodeToolOn and M107 for gcodeToolOff, but I can't find how to raster images.... PWM info is in G1 lines, but this format does not work in Marlin.

Any chance to have some kind of control over how gcode is generated?

jorgerobles, 2017-08-20:
Not really at the moment. There are hooks to append/prepend gcode after
pass but no full blown postprocessor.

tbfleming, 2017-08-21:
The path CAM and the raster CAM are two separate code bases. I wrote the path CAM and later added the options to support non-grbl non-smoothie systems. We tend to avoid touching the raster CAM except when major problems show up. It would be great if someone volunteered to take over the raster CAM; there's a ton of requests for improving it.

openhardwarecoza, 2017-08-21:
@martindb your implementation is "hacking' the fan control to control your laser.  This is not a scenario we want to cater for... adopt a firmware that DOES support laser and you'll be much happier. This is LASERweb, not FANCONTROLweb after all

openhardwarecoza, 2017-08-21:
@tbfleming @lautr3k 
"there's a ton of requests for improving it." - care to put them into an issue / label existing issues so we can talk to @lautr3k and find out what it's going to take (;

universam1, 2017-08-21:
Sorry for my potential ignorance @openhardwarecoza but I can't see where a laser-aware Marlin derivate changes the situation. I'm running latest Marlin where Laser commands are supported (G3-G5) still it is not compatible with the raster CAM! (Which is actually no difference then M106)
So IMHO the Cut and Fill went the right way to split off and make the Laser commands adjustable, I think @tbfleming thats the perfect right way!

I'm bothering myself a long time to switch to GBRL to have proper support, but the truth is that I and more and more are using their machines in a hybrid way as a 3D Printer, Laser and perhaps CNC, so that way I cant switch firmwares all the time.
I wonder how huge the rework is because otherwise it would even make sense to even create a hacky postprocessing convertion script ...

openhardwarecoza, 2017-08-21:
@universam1 : see https://github.com/MKFirmware/MarlinKimbra/blob/V4_2_9/MK/Configuration_Laser.h#L61  which honors the "standard" way with g1 = laser on, g0 = laser off and S = intensity... no need to change anything.  MarlinKimbra, Grbl, Smoothieware all use the same semantic...

universam1, 2017-08-21:
Well I'm kind of sticking to Marlin with my extrem tunings... but I might try to give MarlinKimbra a try. But forgive me, I think it's not the right approach 😃

openhardwarecoza, 2017-08-21:
Approach aside, we lost the developer and its going to be a while... For
something that only affects a couple users with plenty of easy alternatives
(mk etc) you can simply upload.

martindb, 2017-08-21:
I know M106/M107 is not standard for laser control. But as @universam1 said, in a multipurpose machine (laser, 3d print, cnc), Marlin is widely used. Also, Marlin is incorporating more cnc specific gcode in near future. Just I've asked to "customize" the gcode ouput for raster as cut/fill do.

BTW, with a simple postprocessor (awk '{if ($1~/^G1$/ && $4~/^S/) {print "M106 "$4"\n"$1" "$2" "$3} else {print $0}}' input.gcode > output.gcode) I can engrave with LW in my marlin MPCNC and it's great!

universam1, 2017-08-21:
nice script @martindb I was working on a python one but this is better! Thanks will try

openhardwarecoza, 2017-08-21:
Buy @lautr3k some beers over on https://www.paypal.me/skarab and see if you have luck coaching him out of hiding, or find volunteers to take over the Rasterizer.js development (We depend on these https://www.npmjs.com/~skarab )

universam1, 2017-08-21:
Here is the windows postprocessing with powershell
```powershell
gc .\input.gcode|%{if($_-match "(.*)(S\d\.?\d*)"){("M106 {0}`r`nG1 {1}" -f $Matches[2],$Matches[1]).Replace("G1 G1","G1")}else{$_}}
```

Btw. @martindb are you not facing lines without the `G1` prefix?

martindb, 2017-08-21:
If you use vervose gcode, all the lines with S PWM start with a G1 as I can see. Do you have any case this is not ok?

universam1, 2017-08-21:
Good to know haven't tried verbose option.

martindb, 2017-08-21:
Thanks @openhardwarecoza I'll try to talk with him.

prusanoff, 2017-12-12:
Hey Guys,  
     still getting the same problem with the Intensity Settings
Running: 

- MKS Gen Ramps 1.4

- Chinese K40 or a cheap 40 Watt Laser

-  Marlin 1.1.6

In Configuration_adv.h
I set up me 
I turned on the M3 command and I am using pin 4 for PWN and pin 6 for on laser

I tried using this old shitty version of marlin and it sucks like big time
and like @universam1 already said I rather stick to the current version of marlin it has too many improvements to just pass up Plus LW4 has the inputs for laser intensity settings why does it not just work

I'm a node.JS developer with some CNC experience Ill look into the lw.raster-to-gcode.js see what i can hack up

jorgerobles, 2017-12-12:
Thanks!

Just to comment that check for the generated gcode first to discard any
other misconfiguration. Use a simple ramp greyscale bitmap to detect if is
LW gcode generation, or maybe a weird Interaction with marlin, as GRBL is
running ok so far.

Andycart, 2017-12-12:
Hi. I'm running a K40 with MarlinKimbra firmware on a Mega256 and RAMPS. What's the exact response required from M115 to get Laserweb4 to recognise this? I've tried "mk_" and "FIRMWARE_NAME:MK_" but each time I connect LW tells me there is no supported firmware detected and drops the connection?

cprezzi, 2017-12-12:
The server searches for "FIRMWARE_NAME:MK" to detect MarlinKimbra, and then it searches for "mk_" to get the 5 digits version number.

Andycart, 2017-12-12:
Will any version number do? And can the “mk_12345” be anywhere in the string returned by M115. I’m hacking language.h to provide the correct string so an exact, usable example would be really helpful. Many thanks. Andy

cprezzi, 2017-12-13:
Yes, the version number is not important, it's just shown in the connect message. 
The "mk_12345" can be anywhere in the same line as "FIRMWARE_NAME:MK" but not after a line break \n.

cprezzi, 2017-12-13:
There is a sequential firmware detection process which tries to detect the firmware. 

- This process first checks if we get the Grbl welcome string within 1s after connecting. 
- If not, it sends "version" and waits max 1s for the smootieware string. 
- If this is also not detected, it sends "{fb:n}" and waits max 1s for the TinyG string.
- If still not detected it sends "M115" and waits max 7s for the MarlinKinbra string.

The timeouts can be adjusted by environmen variables (in .env file for Windows).

Andycart, 2017-12-13:
Many thanks. It’s working fine now. I’ve done several vector cuts and a raster engrave. All good.

prusanoff, 2017-12-18:
So you guys talked me over to marlin-Kimbra which I have to say is nice however it has some real issues. On tight corners, it almost stops vs on regular marlin flies through tight corners. I love connecting Directly to MK to laser web, however, is I cannot do tight corners how am I supposed to use my laser cutter?

cprezzi, 2017-12-19:
Please check your accelleration and jerk settings. I don't think that MalinKimbra is much different from Marlin (same codebase for moves).

prusanoff, 2017-12-19:
@cprezzi  are you serious that's the first thing I did. Did you even look at the link to the other discussion

jorgerobles, 2018-05-01:
@universam1 please check new marlin flavor gcode generator (Settings gcode, choose marlin generator) https://github.com/LaserWeb/LaserWeb4/pull/473

closed 2018-05-01 by jorgerobles