LaserWeb/LaserWeb4 #396: Unable to clear 'Hold' after 'Abort'

domenic-d, 2017-08-18:
Frontend: 4.0.989
Backend 4.0.115
Smoothie Firmware: Latest CNC edge-049af91

If a job is paused (Hold displayed) and abort is pressed, it is not possible to clear the 'hold' without restarting the controller.

cprezzi, 2017-08-18:
Ok, I'll check that...

cprezzi, 2017-08-23:
The problem is caused by smoothieware, which goes to alarm state on each **ctrl-x** even when not needed (no job running and position save). Grbl does handle that better and goes only to alarm when a move was interrupted.

domenic-d, 2017-08-24:
@cprezzi Are you able to clear that alarm?

cprezzi, 2017-08-25:
Yes, if I press reset alarm (after pause and abort), the alarm goes away. 
Doesn't that work on your machine?

domenic-d, 2017-08-25:
@cprezzi No.

cprezzi, 2017-08-28:
Ok. I was able to replicate the problem with Smoothieware **firmware-cnc-latest.bin**. It works perfect with **firmware-cnc.bin**. Something must have changed on smoothieware side.

domenic-d, 2017-08-29:
I have also tested it on a default smoothie board and it is happening as well. I wanted to ensure it was not specifically our board causing it.

cprezzi, 2017-08-29:
Which Smoothieware version did you use? As I said, I only get the problem with firmware-cnc-latest.bin, but not with firmware-cnc.bin. I was testing with the Cohesion3D Remix.

domenic-d, 2017-08-31:
@cprezzi I used the latest smoothiware CNC version.
After the issues we were having with the older non-CNC firmware's compatibility with LW, we now compile our config into the firmware and hence are always using the latest version.

Can you provide any specific details you need to resolve this so we can perhaps get Jim or Arthur involved to see what has been changed?

cprezzi, 2017-08-31:
I think you compile the same version as the so called **firmware-cnc-latest.bin** on the smoothie page. Can you please try the **firmware-cnc.bin** from https://github.com/Smoothieware/Smoothieware/tree/edge/FirmwareBin, just to confirm my test?

cprezzi, 2017-08-31:
If I know what exactly changed on Smoothieware side (related to hold, resume, ctrl-x and homing) I can adjust my state handling. 
The problem you get is that LW thinks it's still in pause after the abort.

cprezzi, 2017-08-31:
The only concern I have is, that two different behaviours would be needed for users with latest or older firmware.

domenic-d, 2017-08-31:
I have tested it with firmware-cnc.bin and can confirm this works correctly as you have observed.
Pause followed by two aborts places the machine in an 'idle' state correctly. Jobs can then be restarted as expected.

I understand your concern about having the two behaviours. At the end of the day, laserweb should be able to support the 'latest' firmware, just like we experienced recently with it not supporting the older firmware version we were using. This is the same situation with smoothiware users needing to update a while back so that they could use laserweb at all.

cprezzi, 2017-09-01:
For me, latest is something like beta? I always use the normal (not latest) version, because I've read somewhare that lastest could be less stable.

Did you find out what changed in Smoothieware between the two versions?

domenic-d, 2017-09-03:
@cprezzi I didn't pursue what changes because you mentioned you knew exactly what changed.

domenic-d, 2017-09-04:
@wolfmanjm Can you please advise whether we should be working with the 'latest' version of the CNC firmware or is this 'beta', as Claudio is asking?

There has been a change made to the 'latest' firmware which has broken the way Laserweb handles pause/resume/abort.

There is uncertainty as to whether LW should be revise to accomodate this change or users should be restricted to the older version of the firmware.

wolfmanjm, 2017-09-04:
I'll look into it.The old frmware ignored hold !/~, the latest firmware will go into a hold state although not pause immediately.

It is possible that if you abort (control X) while it is in hold it will not clear the hold anymore. If so I will fix it so that an abort will clear any hold.

domenic-d, 2017-09-04:
@wolfmanjm Thanks for the info. That certainly appears to what is going on. Please let us know when we can test your fix.

On a side note, we usually run our segments at 0 or 0.1. With these settings, the delay in pausing is not very long and acceptable in most circumstances.

wolfmanjm, 2017-09-04:
I presume by ABORT you mean KILL (sending control X)? Abort in smoothie is a different thing and stops a job running from the internal SDCard.

wolfmanjm, 2017-09-04:
Looking at the code it should work correctly. Feed hold is cleared regardless of being in HALT mode.
Are you expecting a control X to also clear HOLD? if so that currently does not happen you need to send ~ to clear the HOLD even when in HALT mode.
This is with the latest code, not sure if the prebuilt stuff has this HOLD in it at all.

wolfmanjm, 2017-09-04:
Confirmed it works as I expect it to work..
```
?<Idle|MPos:0.0000,0.0000,0.0000|WPos:0.0000,0.0000,0.0000|F:4000.0000>
!
ok
?<Hold|MPos:0.0000,0.0000,0.0000|WPos:0.0000,0.0000,0.0000|F:4000.0000>
ALARM: Abort during cycle
~
ok
?<Alarm|MPos:0.0000,0.0000,0.0000|WPos:0.0000,0.0000,0.0000|F:4000.0000>
$X
[Caution: Unlocked]
ok
?<Idle|MPos:0.0000,0.0000,0.0000|WPos:0.0000,0.0000,0.0000|F:4000.0000>
```
This shows that I can issue HOLD with a !, then control X to HALT, then issue ~ to clear HOLD, then $X to clear HALT, and we end up back in IDLE, without the ~ we would be back in HOLD.

domenic-d, 2017-09-04:
@wolfmanjm Thanks for the info. @cprezzi oversees this part of the code and I guess he will ask for any clarification if needed.

Just to clear up on the original question, is 'latest' the version users should be running?

wolfmanjm, 2017-09-05:
Yes always use the latest edge.

FabCreator, 2017-09-05:
I thought Laserweb was using CNC version?

cprezzi, 2017-09-05:
@FabCreator Yes, absolutely. On the whole thread we are talking about the CNC firmware.

cprezzi, 2017-09-05:
@domenic-d I will check my code for the procedure like Jim described.

domenic-d, 2017-10-02:
@cprezzi Just checking in to see whether you have had a chance to look into this?

cprezzi, 2017-10-02:
@domenic-d I was very busy last month and forgot about this problem. Sorry!
I did now adjust the process on "clear alarm" for smoothieware (send additional ~after $X). This seems to resolv the problem. It will be published with backend version 126.

domenic-d, 2017-10-02:
@cprezzi Thanks Claudio.
I will check this with the latest version of Smoothieware when its published.

domenic-d, 2017-10-03:
@cprezzi Hi Claudio.
I have just tested the latest smoothie CNC firmware (compiled locally).
It appears that this version allows clearing of alarms correctly.

A new problem is that when connecting via wifi, 'No supported firmware detected' is returned. This error does not occur with the previous version of the firmware. It also does not occur when connecting via USB.

I suspect there may be a problem with the way lw.comms is parsing the smoothie connection string?

I have attached two screen grabs showing USB and WiFi connection responses by the console. Can you provide any information on why this would be happening?

[image]
[image]

cprezzi, 2017-10-04:
@domenic-d Good question. It could be that it just takes a little longer until smoothieware sends the welcome string and this takes more than the default timeout of LW. 
You could expand that timeout by adding "SMOOTHIE_WAIT_TIME=5s" to the ".env" file. (default is 1s)

FabCreator, 2017-12-29:
[image]

Hi All,
During a long engrave we occasional get this issue, the job moves into a Hold state, we are unable to clear the hold and continue the job. Any ideas to why it could be moving into hold? Is there any way we can continue the job without having to abort and then ~ to clear the hold.

Kind regards,
Bonne

FabCreator, 2017-12-29:
[image]

jorgerobles, 2018-05-01:
@FabCreator  Probably gone to Alarm state for a raster issue. There have been some optimizations did by @llamasoft  last week. Will be added next compile so you can test if still happening.

closed 2018-05-01 by jorgerobles