LaserWeb/LaserWeb4 #397: first time clean install using "npm install" fails on "node-gyp rebuild"

DouglasPearless, 2017-08-21:
Running "npm install" on a fresh fork of LaserWeb4 (in preparation to adding some new functionality) fails on the "node-gyp rebuild" step:

```
> buffertools@2.1.6 install /Users/douglaspearless/Documents/workspace/LaserWeb4/node_modules/unicode/node_modules/bufferstream/node_modules/buffertools
> node-gyp rebuild

  CXX(target) Release/obj.target/buffertools/buffertools.o
In file included from ../buffertools.cc:17:
In file included from /Users/douglaspearless/.node-gyp/0.12.3/src/node.h:61:
/Users/douglaspearless/.node-gyp/0.12.3/deps/v8/include/v8.h:5768:54: error: 'CreateHandle' is a protected member of 'v8::HandleScope'
  return Handle<T>(reinterpret_cast<T*>(HandleScope::CreateHandle(
                                        ~~~~~~~~~~~~~^~~~~~~~~~~~
/Users/douglaspearless/.node-gyp/0.12.3/deps/v8/include/v8.h:820:29: note: declared protected here
  static internal::Object** CreateHandle(internal::Isolate* isolate,
                            ^
1 error generated.
make: *** [Release/obj.target/buffertools/buffertools.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:269:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Darwin 16.7.0
gyp ERR! command "node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/douglaspearless/Documents/workspace/LaserWeb4/node_modules/unicode/node_modules/bufferstream/node_modules/buffertools
gyp ERR! node -v v0.12.3
gyp ERR! node-gyp -v v1.0.3
gyp ERR! not ok 
```

Thoughts?

Cheers
Douglas

tbfleming, 2017-08-21:
Which OS and Node version? Windows 10 / Node 6.10.3 don't have this problem.

tbfleming, 2017-08-21:
Just spotted it in the error message:

```
...
gyp ERR! System Darwin 16.7.0
...
gyp ERR! node -v v0.12.3
...
```
That version's too old.

DouglasPearless, 2017-08-21:
Thanks for the pointer, using your pointer I went to the NPM forums and found [https://github.com/npm/npm/issues/17996](https://github.com/npm/npm/issues/17996) and in essence there is a bug that prevents npm upgrading itself and is fixed by:

`curl "https://nodejs.org/dist/latest/node-${VERSION:-$(wget -qO- https://nodejs.org/dist/latest/ | sed -nE 's|.*>node-(.*)\.pkg</a>.*|\1|p')}.pkg" > "$HOME/Downloads/node-latest.pkg" && sudo installer -store -pkg "$HOME/Downloads/node-latest.pkg" -target "/"`

Then the install works for those who have older NPM installations already on their machines.

So I propose to close this issue :-)

closed 2017-08-21 by DouglasPearless