LaserWeb/LaserWeb4 #398: lw.lasterizer improvements

tbfleming, 2017-08-21:
Running list of desired rasterizer improvements as we remember them.

* High priority: Option to vary intensity with speed. Users aren't getting anywhere near the image quality they used to get when this was present.
* Consider dropping the image manipulation options (contrast, etc.). We might just drop it in the UI. Both new users and expert users ignore them; expert users use external apps to prep images. Unknown: do intermediate users use these options?
* Support new settings for LinuxCNC, Mach, and Marlin users (#334).
* Long term: consider adding post processing support. I started experimenting with an interface for this on the new lathe stuff. If it ends up working well I may spread it to the rest of the path CAM. The start of the interface lives at https://github.com/LaserWeb/LaserWeb4/blob/e53c11bc0cfb6d3be940045aeb00a65d57c81d58/src/lib/cam-gcode-lathe.js#L18

openhardwarecoza, 2017-08-21:
"High priority: Option to vary feed with pixel value."!     Yay!  Best news ever!   #BRINGITBACK!

tbfleming, 2017-08-21:
Marking it high priority doesn't mean we have any volunteers to do it :(

domenic-d, 2017-08-25:
The image filtering options became far more useful once the preview became available.

IMHO, the options in order of usefulness and should be considered leaving in are:
#1 Invert Color (very useful)
#2: Brightness/Contrast (covers most of the tweaking necessary before needing to jump into external photo editing software)
#3 Smoothing (would be far more useful if amount could be dialled in)
#4 Dithering (would be useful if this feature was expanded.)

DarkAlchy, 2017-10-06:
I added 2, or 3, lines to Marlin and I have been using LW4 for a few days.  All I had to do was tell it smoothie and verbose output so the g commands are all on one line.  btw, there is a bug in LW4 when I touch the gcode section it loses all LF/CR and there is no way to bring it back so I am stuck at one line.  I don't do github so opening tickets etc... I have no idea and this may already be a known issue.

tbfleming, 2017-10-06:
> when I touch the gcode section it loses all LF/CR

What do you mean?

DarkAlchy, 2017-10-06:
Well, if I add more than one line or if I even touch the gcode already there it will force that gcode section on one line so I will give you an example.  Settings->Gcode->Gcode Start->
G21         ; Set units to mm
G90         ; Absolute positioning

If I add to that or modify that (say I erase the g and put it back) in the actual gcode written it loses the LF/CR so becomes G21         ; Set units to mmG90         ; Absolute positioning

If I go to any section in there to add my own code if I do more than one line it does the same as I just showed.  I thought I had to add \n but that didn't do anything but at it between the last letter of the line and the first letter of the next line but all on one line.  The only way to fix it once that happens is to delete the machine and start over.

jorgerobles, 2017-10-06:
Could be it only loses CR? Could you check with an editor that supports LN
format? Scite, vscode, etc..

DarkAlchy, 2017-10-06:
I will go one better and use a hex editor on it and see what it is and what it becomes.

DarkAlchy, 2017-10-06:
0x0d,0x0a becomes 0x0A so the CR is lost.

jorgerobles, 2017-10-06:
Then should not be a problem running the gcode. CR is a win only
visualization issue.

DarkAlchy, 2017-10-06:
Well, easy enough for it to be fixed and I am unsure if Marlin will still work like that or not.  Maybe but it still sucks to see it all on one line be it just a visual bug or not.  I hope it gets fixed since it should be dead easy to take care of it.  I can't open Marlin to see if it wants 0a or 0d or 0d0a and since Marlin support is in the works best to nail it down.

jorgerobles, 2017-10-06:
Yes, marlin works.. mine at least. got a couple hundred gcode files on my
3d printer that way.

DarkAlchy, 2017-10-06:
You mean the Marlin code for LW4 has been implemented now?

jorgerobles, 2017-10-06:
No, I mean marlin firmware understand LN only files :)

DarkAlchy, 2017-10-06:
Ahhh, I figured but wasn't sure.  As I said I hope this visual bug gets fixed but I wonder why modifying the boxes eats the CR?

jorgerobles, 2017-10-06:
not sure why. All the devs use Linux/Mac machines, so could have been
overlooked.

DarkAlchy, 2017-10-06:
Yep, as I figured.  Well, hopefully someone reports it and puts in a ticket for the issue.

cprezzi, 2017-10-06:
@DarkAlchy The lw.comm-server only supports MarlinKimbra yet, because MK supports S values in G moves. Without that (old Marlin), the axes moves would stop at each S change, which makes it unusabe for raster engraving.

By the way: The multiline gcode settings fields work like expected on my Win10 machine.

tbfleming, 2017-10-06:
@DarkAlchy Are you using Notepad to view the gcode? Almost every other editor for Windows doesn't care about CR. I use this from Microsoft: https://code.visualstudio.com/ .

DarkAlchy, 2017-10-06:
@cprezzi I added the S command to vanilla Marlin so it works as I didn't want to mess with any forks since it is a 3d printer machine mainly.

@tbfleming  Yep, been using it for 5 years for files like this as wordpad I never liked and notepad++ is way overkill.  Thanks for the link.