LaserWeb/LaserWeb4 #399: Switch from npm to yarn?

tbfleming, 2017-08-22:
Think of yarn as a faster version of npm, with sane status and error messages. @openhardwarecoza : it has bandwidth saving features.

To try it:
```
npm i -g yarn
cd LaserWeb4
git checkout yarn
yarn run installdev
yarn start
```

To see where it really shines, do this after you did the above:
* wipe node_modules directory
* ```yarn``` (equivalent to ```npm install```)

See the chart at https://yarnpkg.com/en/docs/migrating-from-npm#toc-cli-commands-comparison for the commands to replace the npm commands.

kaosat-dev, 2017-08-27:
Most of the features that were specific to yarn have been added to the latest npm (v5)  (including lockfile etc), so not 100% sure how worthwhile a switch might be. Just my 2 cents though ;-)

jorgerobles, 2018-05-01:
This is already done isn't it?

tbfleming, 2018-05-01:
Only on the trial branch