LaserWeb/LaserWeb4 #400: The "Home all" button fails to bring the head to the origin of the laser cutter

kcleung, 2017-08-22:
### Follow this steps to open an issue ###

- [ ] If refers software, include `Workspace.json`.  This is available on *CAM tab top, Workspace -> Save*. Prepare your work, save `workspace.json`, rename as `workspace.json.txt` and attach in this issue. *All resources regarding the issue should be attached (SVG files, images, etc.). An issue opened without resources will be automatically closed upon inspection*

- [ ] If refers hardware, include enough images, log files, and optionally videos (or link to a Cloud Drive folder) to inspect.
[Laserweb-Workspace.json.txt](https://github.com/LaserWeb/LaserWeb4/files/1243631/Laserweb-Workspace.json.txt)

I am using LaserWeb4 on an OS X 10.12 to control the Emblaser 2 laser cutter.  When I click the "Home all" button, the head travels to the lengths of the Y then X axis, but fails to go back to the origin position marked on the machine (bottom left corner).

cprezzi, 2017-08-23:
@domenic-d I think this is a question for you.

domenic-d, 2017-08-25:
@kcleung Please raise a support ticket on our Help Centre for all Emblaser related issues.
https://darklylabs.zendesk.com/hc/en-us

closed 2017-08-25 by domenic-d

kcleung, 2017-08-26:
@domenic-d I have already raised a support ticket (#858) at your site, please have a look.