LaserWeb/LaserWeb4 #401: NOT AN ISSUE - PAGE BEING USED IN AN IMAGES TEST FOR WIKI

funinthefalls, 2017-08-29:
[image]

openhardwarecoza, 2017-08-29:
So CNCPro.co now?  <frown>

funinthefalls, 2017-08-29:
[
[image]
](url)

openhardwarecoza, 2017-08-29:
Anthony... this sort of thing is WHY i kicked you out of doing docs.  Seriously WTF.   We moved from Github wikis, to a proper PAID FOR hosting with proper CMS so we DONT HAVE TO RESORT TO STUPID HACKS like this!

openhardwarecoza, 2017-08-29:
Also, anyone care to remember WHY we went from Git wikis to CMS?   for the sake of USERS: Git wikis are too limited to do proper menu structures and search

openhardwarecoza, 2017-08-29:
This is another example of the shit you did in the old days too: nonsensical shit is worth less than NO docs at all... 

Who uses curly brackets as a whatever it was supposed to be
[image]

firmware-cnc.bin 
[image]

My point is you need someone who KNOWS the app, to dopcument the APP, not some bored guy who thinks he has a clue

funinthefalls, 2017-08-29:
As always, you are entitled to you opinion. I was just doing what Todd asked me to do. If you have an issue with the documents being done this way I suggest you have a discussion with Todd, Claudio and Jorges. I am nothing more than a volunteer trying to move this project forward. I find your ability to -overreact to things, well frankly I find it as being nothing more than a childish tantrum.

openhardwarecoza, 2017-08-29:
@tbfleming ? @cprezzi @jorgerobles

openhardwarecoza, 2017-08-29:
showing users screenshots off the live version (as per your first post)
leads to more confusion too... at least crop out the browser ffs!

funinthefalls, 2017-08-29:
That screen shot is direct fro CNCPro.co, funny you never has an issue with it when it was posted by someone else, but now that I post it it you comment that it is idiotic. Make up your mind Peter, can't have it both ways.

openhardwarecoza, 2017-08-29:
Then FIX it dont just blindly move existing contents. That's not what was
needed ever

funinthefalls, 2017-08-29:
Fixing is what this process is all about. A message was posted on the LaserWeb g+ community stating it would be a few weeks before everything was moved and edited.

openhardwarecoza, 2017-08-29:
So, do it twice? Upload wrong then te upload correct. Oh all the while
still in the wrong place. Genius! That makes a lot of no sense. Sure you
dont want to go volunteer somewhere else?

funinthefalls, 2017-08-29:
It may be YOUR idea of being the wrong place, but it is where the CURRENT devs wanted me to put it. You can try and bait me all you want Peter, the only thing you will accomplish is creating tons of screen shots that document your ignorance and lack of respect for others.

openhardwarecoza, 2017-08-29:
:) yay more screenshots you wanna forward to everyone via email. Again.
Surprised i dont like you? Whose childish now. Anyway. Fuck forth.

jorgerobles, 2017-08-29:
Please Stop. I think this is not the place to air up personal grudges.

Just ask @tbfleming  politely what all of this intended (I've been away for a while, and don't know about that conversation).

I believe in people's good will.
Disabling wiki until that is a good option.

tbfleming, 2017-08-29:
I'm pissed off at you, @openhardwarecoza . I need to back away before I do something rash.

openhardwarecoza, 2017-08-29:
Thats fine. Just remember the history you  don't even know

tbfleming, 2017-08-29:
I don't give a crap about past history I don't know of, except when you use it as an excuse to go off the rails at someone.

funinthefalls, 2017-08-29:
This thread is still for images destined for use in the LaserWeb wiki. Please do not delete or modify.

cprezzi, 2017-08-29:
@openhardwarecoza What did you make with cncpro.co??? I spent many hours for cncpro.co!! 
The wiki was not ment to replace cncpro.co, it was ment to give impatient users a quick entry!

Please restore cncpro.co.

funinthefalls, 2017-08-29:
I am closing this issue as it strayed from its purpose. I will be making another issue to handle the image uploading on a temporary basis

closed 2017-08-29 by funinthefalls

cprezzi, 2019-12-17:
[image]