LaserWeb/LaserWeb4 #403: ##NOT AN ISSUE - FOR UPLOAD OF WIKI IMAGES ONLY - DO NOT USE OR RRESPOND PLEASE**

funinthefalls, 2017-08-29:


funinthefalls, 2017-08-29:
[image]
[image]
[image]
[image]
[image]

cprezzi, 2017-08-30:
@funinthefalls Please don't use your "UFO" Logo on our page. That's nothing official and in my eye, it's ugly. Also, don't post anything about gerbil, it's nothing we officially support.

cprezzi, 2017-08-31:
[image]

[image]

[image]

[image]

jorgerobles, 2017-10-08:
@funinthefalls  is this issue still used? How is the documentation on wiki going?

cprezzi, 2017-10-08:
@jorgerobles This issue is permanently needed. as the pictures in the wiki are related to the pictures in this issue.

jorgerobles, 2017-10-08:
Ok, just curiosity: are the url references not working if issue is closed? We can always reopen or extend with content even closed

cprezzi, 2017-10-08:
I think it should still be working when the issue is closed. I'll test it.

closed 2017-10-08 by cprezzi