LaserWeb/LaserWeb4 #404: CNC-pro.co gone?

FabCreator, 2017-08-30:
any reason cncpro.co is retired?

jorgerobles, 2017-08-30:
🤦‍♂️

arthurwolf, 2017-08-30:
I was about to say : if you want you can choose a domain and I'll buy it and host whatever you want on it, my treat.
However I don't think it'd be a good thing for laserweb to depend on my company's good will to renew it's domain every year.

What we can do however, which would guarantee independence, is that a laserweb contributor takes responsibility for ordering the domain and hosting, and my company pays let's say 5 years of that in advance ( directly to the contributor, we don't have any control after that )
I'm ready to do that if a volunteer pops up, and let's say if one of the two laser cutter companies closely tied to lw are ready to split the cost with us ( robotseed ). Splitting three ways would be even better.
@FabCreator @DarklyLabs ?

I may be wrong and maybe you don't need a domain or hosting at all. In that case just ignore me. But if you do, this might be a solution.

FabCreator, 2017-08-30:
@arthurwolf smoothie has some of the best documentation i have come across, 
What do you use for it? and what does it cost?
I dont think the Wiki on github is very user friendly or powerful
I am up to splitting the costs i guess it depends what it costs in the first place ;)

arthurwolf, 2017-08-30:
We used to use wikidot which is great when starting out, but as the wiki gets better it's limitations are felt ( limits on custom html, css and js ).
So we switched a year back to self-hosting ( on a dedicated server, though a shared one would fit ) and use dokuwiki. We also created a custom "skin" for dokuwiki based on bootstrap and the general previous smoothie look and feel.
I would recommend this solution, but I think in the end, what's best is what is confortable to whomever will be the most active contributor the the documentation ... ( still best to avoid things that really won't be a good fit like mediawiki )

About costs, a domain is like $10/year, and a base dedicated server is $10/month, so $130/year, 5 years is 650, shared between 3 "sponsor" companies is 216. I'm pretty sure 5 years from now laserweb will be able to pay for it's own hosting so this ensures independence forever.

tbfleming, 2017-08-30:
@funinthefalls is porting the existing docs to GH wiki and cleaning them up. He has limited time, so may appreciate additional help. Once that's done we could convert to wikidot or any other wiki or markdown hosting. One option is to use gh-pages. It'd give us full control over formatting since we could also use html+css+scripts. The upside is free hosting and it'd be using something the devs are already familiar with (git). The downside is any doc contributors would have to learn git.

arthurwolf, 2017-08-30:
gh-pages is a really great option, completely recommend that, I just didn't
have time to learn but it's my gold standard. avoids having to pay for
hosting too.

FabCreator, 2017-08-30:
https://wiki.js.org/ just googling, this looks like its powered by Git, might be able to provide a little more control over how to present?

tbfleming, 2017-08-30:
At one point we started to think about forming a management board. We never got around to it, but we probably should to at least own the domain.

cprezzi, 2017-08-30:
Last Year we decided to build cncpro.co because the wiki is too limited with navigation and illustrations and the markup "language" is a unconvenient technology of the last century. Today we have good wisiwyg online editors in cms systems. My concern is, that we would be too much limited by a wiki system. What if we want to add a newslerrter tool, a support ticket system or a cloased user area?

I do also think it's important that the domain is owned by a project management board. Also the hosting should be "owned" by this board.

My company is specialized in CMS solutions (especially with Joomla), and we also develop customer components for Joomla. I can offer setup and maintenance of a Joomla system, if that's what the board wants.

arthurwolf, 2017-08-30:
Modern wiki tools like dokuwiki have all the bells and whistles of a modern
CMS like wysiwig, newsletter etc. And they are wiki so they are great if
you are counting on the community writing the documentation
Obviously, if your company does joomla, everything is going to look like a
nail :) As I said I think the best tool is the one the contributors are the
most familiar with. Just have to figure out what that is.

cprezzi, 2017-08-30:
Did you check the dokuwiki templates? They all look the same, very old-fashioned and limited. 
A modern CMS (Drupal, Joomla, Typo3, even Wordpress) has much more features than a wiki and the appearance can be completely different (depending on the template and module positioning). Just my opinion.

arthurwolf, 2017-08-30:
> Did you check the dokuwiki templates? They all look the same, very
> old-fashioned and limited.

I got a boostrap template and started from there, but as far as pre-made
templates I found there were a lot.

> A modern CMS (Drupal, Joomla, Typo3, even Wordpress) has much more
> features than a wiki
Dokuwiki has a ton of plugins, so far I haven't found anything I wanted it
to do that it didn't.

> and the appearance can be completely different (depending on the template
> and module positioning). Just my opinion.

You are writing documentation. The system has to dissapear to only show the
content, as much as possible. I find wikis are really good at doing that
out of the box.

cprezzi, 2017-08-30:
It doesn't have to be Joomla. I would prefer the gh-pages path over a dokuwiki. It's more streamlined to the way we work anyways.

arthurwolf, 2017-08-30:
Yep, definitely gh-pages is a great solution. Also users don't necessarily
have to learn git : github lets users edit files from the website directly
now.

tbfleming, 2017-08-30:
> github lets users edit files from the website directly now

Ooooh! Do you have a link? My google-foo failed me.

cprezzi, 2017-08-30:
The GH-pages documentation is on https://pages.github.com

tbfleming, 2017-08-30:
First place I looked :). Didn't see the new feature that allows users to edit from the website.

cprezzi, 2017-08-30:
Found a specialized template for gh-pages documentation: http://idratherbewriting.com/documentation-theme-jekyll/index.html

arthurwolf, 2017-08-30:
https://github.com/Smoothieware/Smoothieware/blob/edge/fast-stream.py
click the pen just above the first line of the file

arthurwolf, 2017-08-30:
> First place I looked :). Didn't see the new feature that allows users to
> edit from the website.

It lets you edit from the github repo. Since most gh-page-type sites are
script-generated it'd be easy to add a link to the source of each page ( or
directly to the edit link ) somewhere on each page. Essentially making it a
wiki.

tbfleming, 2017-08-30:
Clever use of an old feature :). I thought you meant it could edit directly on the hosted site.

cprezzi, 2017-08-30:
I was also understanding that **frontend-editing** is possible, which would be very convenient.

domenic-d, 2017-08-31:
Darkly Labs would also certainly be up for splitting the costs to secure a more permanent solution.

cprezzi, 2017-08-31:
For the moment, I have added a custom sidebar to the Wiki, so we could at least make a ordered menu structure.

funinthefalls, 2017-08-31:
Thanks Claudio for the sidebar.

JTechPhotonics, 2017-08-31:
We could help and maybe even host the site...  It just depends on what you guys want to do with it  and if you want it to be associated with us.  We support many other open source platforms including GRBL, so it would not be too different. On our site, we use wordpress, which is pretty extensible and easy to manage multiple users.  I can either put it on our site as a separate area or we can make a whole new site/domain for it.  We were planning on using Laserweb as a solution to our new machine so we will have tutorials and such on our site anyway.  Just let me know-  Jay

cojarbi, 2017-09-01:
Jay

Welcome. Nice to see you finally made it here :)

tbfleming, 2017-09-02:
I just replaced the old logo with a temp one in the GH org and in the community. I also purged the cncpro.co from the GH and community descriptions.

tbfleming, 2017-09-02:
Anyone notice this issue number?

funinthefalls, 2017-09-02:
Not until you just mentioned it.

cprezzi, 2017-09-02:
I have restored the cncpro.co webpage on one of our servers: https://cncpro.yurl.ch (so we have time to decide how we want to solve it in the future)

techpad-marcb, 2017-09-10:
I've just started using LaserWeb and found its a fantastic application. I'd love to contribute in anyway possible. This web hosting and domain issue sounds right up my ally and i'd be willing to work with the team on getting what even is needed setup to support LaserWeb.

closed 2017-10-21 by jorgerobles