LaserWeb/LaserWeb4 #406: G1 code generated fails on M3 command with GRBL_V1.1f_08-01-2017

dutchpatriot, 2017-09-01:
When code generated for Laser cut G1 code generated by Laserweb this doesn't work on above stated version of GRBL.

M3 command will work but defined feedrate will be ignored and drop down to 0 but still move.

G2 and G3 generated code has no problems but this is not (yet) implemented in laserweb4

chamnit, 2017-09-01:
Resolved duplicate issue at Grbl site. Program had gcode commands of 0.01-0.02mm in length. Asking the AVR processor to execute up to 5000 lines/sec. G2/3 arcs program works only because Grbl internally generated more proper line lengths at 0.1-1.0mm.

closed 2017-09-01 by chamnit

dutchpatriot, 2017-09-01:
@chamnit And thanks again, just wanted to remove this...