LaserWeb/LaserWeb4 #407: Node Memory Settings

glennswest, 2017-09-02:
Using the latest build 4.0.989-115-x64 on windows 10 (And i've seen it on mac as well)
If I load a image and its close to my darkly em2 max size it universally seems to result in the part
of the gui blowing up, same image resized works fine. I believe the nodejs memory settings are
not big enough, can we get them set 4x bigger. 

I'd love to build from source, but its not clear what the windows build dependencies are,
can someone point me to a list/script etc for setting up development environment.

I dont mind commiting back/PR

cprezzi, 2017-09-02:
All the GUI stuff is running in the browser (chrome inside the electron exe), not in the nodejs server. The server only needs memory for the gcode while you run the job from the controll tab.

@jorgerobles @tbfleming Any limitations for file sizes in the frontend?

jorgerobles, 2017-09-02:
None than I'm aware of, but I suppose there are.

jorgerobles, 2017-09-02:
Glenn please attach file for review..

Daniel-dk, 2017-09-05:
I think I'm running into the same issue, 

When I load a fairly large image, GCODE generation for engraving gets to around 6 % then the GUI goes blank ( white screen, and my machine's fans slow down again, I it might have crashed or stopped entirely )
[image]

cojarbi, 2017-09-05:
I just downloaded your image and was able to generate the raster code in the than a minute with no issues. I presume machine specs has something to with this. I'm running a surface pro 
[image]

Daniel-dk, 2017-09-05:
mm, interesting, 
The host machine is quite beefy ( i7 with 16 GB RAM, OS: Win10 x64, 500 GB free on main HDD )

Is there a debug or error log that I could analyse to see what might have gone wrong ?

 ( the plan is to use ths laser engraving option with a low power UV laser to make cyanotype images )

Daniel-dk, 2017-09-05:
Laser diameter was set to 0.01 mm, 
When set to 0.1 mm it generates Gcode for the image sucessfully

Might be an issue with the number of instructione generated with such a small increment.

tbfleming, 2017-09-05:
.01 mm would cause it to try to generate ~(200/.01)*(200/.01) = 400,000,000 lines of gcode. That's too big.

cprezzi, 2017-09-05:
The real beam diameter is mostly >0.1mm (if good focus). Configuring anything less that that would just overlap the dots.

closed 2017-10-02 by cprezzi