LaserWeb/LaserWeb4 #408: Wiki discussion

tbfleming, 2017-09-02:
Creating a separate thread for discussion about short-term wiki details, so we can keep #404 about long-term direction.

tbfleming, 2017-09-02:
I just renamed the files to drop ```:/}```. Colons are nice, but don't play well on Windows filesystems. The ```}``` was a temporary workaround. For now we'll need to live with no special characters in titles since the wiki ties them to filenames.

Should we keep the numbers? Maintaining numbers seems like it will get too painful.

cprezzi, 2017-09-02:
The numbers are just one way to order the default pages list. The are not needed for the custom sidebar, but the sidebar needs to be adjusted to every title change.