LaserWeb/LaserWeb4 #410: Unable to start server

domenic-d, 2017-09-06:
We have a situation where we cannot start the Laserweb Server.
O/S = Win10
LW = 4.0.989, 4.0.115

Machine is being recognised in windows, com5 (checked under Device Manager).
Laserweb shows no ports listed.
Server Connection menu shows server disconnected.
Trying to connect does not establish connection.

Any ideas on what may be causing this?
Any suggestions on where we can check further into a cause?

cprezzi, 2017-09-07:
Is something else blocking port 8080? You could activate a server logfile by adding the line `LOG_LEVEL=3` to the `.env` file in the installation folder (on Windows). This writes a logfile.txt in the installation folder.

domenic-d, 2017-10-18:
User resolved this by using a different network within their facility. They were not able to provide any specific reason but as you suggested, it appears to have been a port blocking issue.

closed 2017-10-18 by domenic-d