LaserWeb/LaserWeb4 #411: Buildbotics integration

jcoffland, 2017-09-06:
Would love to integrate LaserWeb4 on the Buildbotics CNC controller.  It has a built-in Raspberry Pi 3 so it could serve LaserWeb over the network.  It would not take much effort to connect it up to the Buildbotics control firmware.  https://github.com/buildbotics/bbctrl-firmware

cprezzi, 2017-09-11:
Just follow the installation instructions on https://cncpro.yurl.ch/documentation/installation/36-install-raspberry-pi and can connect the frontend over Raspi-IP:8000.

jcoffland, 2017-09-12:
I don't think that's quite what we want to do.  I would like to serve the HTML, Javascript and CSS parts of LaserWeb4 from the RPi and have our Python based server code act as the "serial port server" on the backend.  Do you think this could work?  Am I missing anything?

jorgerobles, 2017-09-12:
Unless your python code replicates lw server api, you will still need lw
server to do connection :|

jcoffland, 2017-09-12:
What does the LW server do?  Is it more than just a serial port server?  Can you point me to docs on the API?

jorgerobles, 2017-09-12:
Lw comms server provides an abstraction layer for the hardware,

Check https://github.com/LaserWeb/lw.comm-server

tbfleming, 2017-09-12:
Note that the API may change: https://github.com/LaserWeb/lw.comm-server/issues/47

jcoffland, 2017-09-12:
Our Python code already contains a WebSocket server.  I wonder if we couldn't just implement the API directly.  Our server does not need to support TinyG, GRBL or Smoothie.  Its only purpose would be to support the Buildbotics controller.  I found the [API docs](https://github.com/LaserWeb/lw.comm-server/wiki/API-(for-client-communication)).  It doesn't look too difficult but it would be nice to have some more details on the API.  I.e. exactly what the data format should look like.

tbfleming, 2017-09-12:
I thought the Buildbotics controller was TinyG.

jcoffland, 2017-09-12:
If I had a $ for from everyone that thought it was a TinyG in a fancy box our Kickstarter would be funded already.  We did start out with the TinyG firmware but we've made massive improvements and changes.  The API between the RPi and the AVR is similar to the TinyG but there's a lot more to our controller.  We have a Python program which runs on the RPi which adds a lot of functionality.  I need to have that code in the loop.  For this reason, the interface to LaserWeb would be very different.

cprezzi, 2017-09-12:
@jcoffland You will have to check the server source for all websocket events and emits. I did not plan to document the whole API in detail (too much time consuming).

jcoffland, 2017-09-12:
Perhaps I can help with documenting the API.  It appears to be JSON data in both directions.  It would be nice to have something like this: http://dev.buildbotics.com/ for all the API entry points.  The important data is:

 * Request method (HTTP method, Websocket, etc.)
 * Request format
 * Response format

How do you associate requests with responses over Websocket?

tbfleming, 2017-09-12:
> How do you associate requests with responses over Websocket?

It doesn't. In most cases it broadcasts changes to all connected clients ```io.sockets.emit(...)```. It sometimes sends changes to a single client, but there's not a good way to associate it with a specific request.

Something to watch out for: it uses the ```socket.io``` package, which provides its own versioned protocol. ```socket.io``` clients and servers refuse to talk to each other if the ```socket.io``` protocol version don't exactly match. The server sends the package source to the client to keep the two sides in sync. All this is under the control of ```socket.io```. I hate it and hope to eventually dump it.

cprezzi, 2017-09-13:
Sorry, but I love the way socket.io works and I don't see any reason to replace it.

cprezzi, 2017-09-13:
I hate react/redux, but I do not try to replace it ;)

tbfleming, 2017-09-13:
socket.io stops other languages from talking to the server.

jcoffland, 2017-09-13:
I had the same problem with _socket.io_.  I ended up switching to [SockJS](https://github.com/sockjs/sockjs-client) because I couldn't get _socket.io_ to work with Python.  On the Python side, I'm using _tornado_ and _sockjs.tornado_.

Haschtl, 2019-07-07:
I have a working example of the Python3-socketio implementation for communication with LaserWeb4.
[https://gist.github.com/Haschtl/e4953ca8a15c53a079d871fa74a03f68](url)

It's working quite good. You may not need the functions at the end of the API. I just made these to have the same data-structure as my Octoprint-PythonAPI, because i'm using both on one RPi

cprezzi, 2019-07-08:
@Haschtl Nice stuff, appreciate your work!

Haschtl, 2019-07-08:
Thanks! I had some errors in the PythonAPI, just fixed it.