LaserWeb/LaserWeb4 #413: No preview with Linux bundle

tbaumann, 2017-09-18:
Hi,

I hope you guys can help me figure out what is going on. I got Laserweb with grbl and an EleksLaser pretty much running. But the canvas preview is not working right.
 Frontend: 4.0.989
 Backend: 4.0.115
Downloaded as binary and running on Ubuntu Xenial.

The problem is the following. When I add documents to the workspace, I don't see any preview. Neither from the document itself (raster and svg) nor from the generated gcode.
All the bits in the left tab work fine. I can drag and drop documents into the gcode generation queue.
The produced gcode executes fine.

A few other things on the canvas to the right work too. Raster preview, webcam and this little scaling tool.

I'm using Chrome directly. But even if I use the browser shipped in the app-image it's the same thing.
Chrome version 60.0.3112.101 (Official Build) (64-bit)

When I open the page in Firefox and I see a very different picture. All the drag and drop stuff doesn't work. But the canvas doesn't just have scaling axis, it also has a line grid. And the laser dot is shown and images of documents are shown when I add them.

I guess what I should be seeing is somewhere in-between.

Resources:
[laserweb-settings.json.txt](https://github.com/LaserWeb/LaserWeb4/files/1311928/laserweb-settings.json.txt)
[image]
[Chemex-light.svg.txt](https://github.com/LaserWeb/LaserWeb4/files/1311935/Chemex-light.svg.txt)
[image]
[image]

PS: The generated gcode would not light the laser. I had to add M3 as TOOL ON code. Makes sense?

jorgerobles, 2017-09-22:
First screenshot is related to no opengl or another kind of hard breaking error. 
Firefox is not a supported browser.

>PS: The generated gcode would not light the laser. I had to add M3 as TOOL ON code. Makes sense?

Yep. You should customize your gcode generation in settings if is needing any special code to be used to make your machine work.

gfwilliams, 2017-09-23:
I have the same here. I'm just using the app bundle - I don't see grid lines or the preview itself, but I do see text - and the rest of LaserWeb works great.

Looking at the devtools I see `WebGL: CONTEXT_LOST_WEBGL: loseContext: context lost` as a warning, but that appears to be the only GL error.

However refreshing the windows gives `canvas.getContext('webgl', {...}) returned null` as an error and the main window shows `OpenGL won't start. This app can't run without it.`

Any ideas what we can do to debug this? I definitely have OpenGL support, and WebGL on normal Chrome and Chromium works great - I've used it on several sites without issues.

gfwilliams, 2017-09-23:
Same issue when connecting with Chrome - and I can open another tab and use WebGL sites just fine. I'm using a standard Intel Kaby Lake for graphics.

tbaumann, 2017-09-23:
That is the crazy thing. chrome://gpu/ is mostly green for me.
[image]
The only thing I don't have is some hardware acceleration and experimental features.
http://webglreport.com/ is also all green.

Chrome console log:
[image]

Is there any way we can debug this?

jorgerobles, 2017-09-23:
I've got it running on an asus Rog with ubuntu mate... :{

tbaumann, 2017-09-23:
https://www.khronos.org/webgl/wiki/HandlingContextLost perhaps relevant?

gfwilliams, 2017-09-23:
@jorgerobles which OS are you using? It looks like the Asus ROG has NVidia graphics, which might explain it.

It could be something about the Intel graphics driver that causes the context to be lost at times?

jorgerobles, 2017-09-23:
It's an ubuntu mate 15 i think
 Nvidia 980m

gfwilliams, 2017-09-23:
I've been trying to get chrome to do software rendering, but without success so far. However starting Chrome from the CLI does yield the following:

```
[20836:20836:0923/150104.670113:ERROR:gles2_cmd_decoder.cc(16034)] Offscreen context lost via ARB/EXT_robustness. Reset status = GL_INNOCENT_CONTEXT_RESET_KHR
[20836:20836:0923/150104.670330:ERROR:gles2_cmd_decoder.cc(4326)]   GLES2DecoderImpl: Context reset detected after MakeCurrent.
[20836:20836:0923/150104.670579:ERROR:gpu_channel_manager.cc(188)] Exiting GPU process because some drivers cannot recover from problems.
[20836:20836:0923/150104.670616:ERROR:gpu_channel_manager.cc(188)] Exiting GPU process because some drivers cannot recover from problems.
[20987:20987:0923/150104.787787:ERROR:gles2_cmd_decoder.cc(17777)] [.DisplayCompositor-0x3c4e887fa800]GL ERROR :GL_INVALID_OPERATION : glCreateAndConsumeTextureCHROMIUM: invalid mailbox name
[20987:20987:0923/150104.787823:ERROR:gles2_cmd_decoder.cc(9787)] [.DisplayCompositor-0x3c4e887fa800]RENDER WARNING: texture bound to texture unit 0 is not renderable. It maybe non-power-of-2 and have incompatible texture filtering.
```

So it looks vaguely expected, and could just be that the context (incl textures) is getting lost because Intel's driver isn't as careful as NVidia's. 

This is the first time I've looked at LaserWeb's source so I'm not 100% sure where I should be looking. However it seems like there's all the code in there so assuming the context lost handler is added it shouldn't be too hard to iterate through the loaded document(s?), delete the texture handles, and call 'update' on them again.

jorgerobles, 2017-09-23:
@tbfleming ?

tbfleming, 2017-09-23:
I've never seen a case where software rasterization works. 

[image]

WebGL is effectively dead without this.

jorgerobles, 2017-09-26:
Weird, my setup reports similar, but works

[image]

tbfleming, 2017-09-26:
Odd. Maybe that setting's poorly named. Rasterization in GPUs normally means turning triangles and other primitives into fragments (pixels), a central part of the GPU pipeline.

gfwilliams, 2017-09-26:
@jorgerobles do you have a way of testing with an intel gpu on linux? seems like the intel driver drops the context, which causes these issues.

If someone could give me some pointers on how to force textures/buffers to be reloaded then I could test here?

tbfleming, 2017-09-26:
It tears down and recreates the WebGL context when components/workspace.js is hotloaded. hotloading isn't part of the electron builds. To get that file to hotload:
* use ```npm run start```
* make a non-whitespace change to workspace.js. e.g. add or remove a console.log(...).

tbaumann, 2017-09-26:
I went to great lengths to keep the intel driver and not run the Nvidia card. (Power consumption and quite frankly quality of the drivers)
Theoretically I could find out how to switch my system to the Nvidia drivers. But I suppose we have pretty much validated it's a intel specific issue, or perhaps one that is more likely to be triggered with intel.
But let me know if I can somehow help testing anything.

Thanks guys

gfwilliams, 2017-10-16:
Ok, I just did as suggested with the non-whitespace change to `components/workspace.js` - I get an `OpenGL won't start. This app can't run without it.` message.

However, I also get `Rats! WebGL hit a snag` from Chrome, followed by my window manager crashing. Yay Intel!

So at this point I'm thinking Intel's drivers for Kaby Lake (and others?) on Linux are full of bees, and it's probably time to switch back to something with dedicated graphics. The mild annoyance here is that even if I used a Pi as the server, because it's client side I can't actually use my Laptop.

I'd suggest closing this then - it looks like it's not specifically LaserWeb related.

I actually had issues on an older Linux & Intel laptop with a different (native) CAD program (Repsnapper iirc). I guess the GL_LINES code paths are used reasonably rarely so are probably pretty buggy.

closed 2017-10-21 by jorgerobles

gfwilliams, 2017-10-23:
Just to add - this is definitely an Intel driver issue. If you look in `dmesg` after this happens you'll see a message indicating that *the GPU itself had crashed and had to be restarted* \o/

I'm trying to look into this, but if not now then in the near future I imagine updating to the absolute latest Intel XOrg drivers will fix this problem.

kennpie, 2020-04-29:
I'm having the same issue: no image preview in LW4, no gcode graphic preview, no "burn" graphics or simulation... but everything else works, including lasering.

older Toshiba Satellite laptop, AMD Athlon X2 64, ATI Radeon graphics, Ubuntu 14.04LTS
Eleksmaker A5 with grbl 1.1
lw.comm-server-4.0.130-x86_64.AppImage

cprezzi, 2020-04-29:
@kennpie How did you install LW4 on Ubuntu?

kennpie, 2020-04-29:
@cprezzi - thanks for the FAST response!  I'll give that a try.