LaserWeb/LaserWeb4 #414: [Feature Request] Better Webcam support including timelapse function

mrcg, 2017-09-22:
I would like to ask for better webcam support so that the webcam live stream can be easly expanded in control/move mode and while engraving activity.
Beside this i would like to have automatic timelapse generation while engraving activity (user to set amount of pictures every X seconds).
More or less exactly the same implementation like done in Octoprint.

https://github.com/foosel/OctoPrint

jorgerobles, 2017-09-22:
> I would like to ask for better webcam support so that the webcam live stream can be easly expanded in control/move mode and while engraving activity.

You know that you can spawn a floating window on LW, that can me moved around and be resized?

[image]

> Beside this i would like to have automatic timelapse generation while engraving activity (user to set amount of pictures every X seconds).

That is unlikely to be done on LW part, because cannot be done with frontend technologies in a performant way. An alternative could be using an external ipcam software to make capturing.

:|

mrcg, 2017-09-22:
I really like the implementation as done in Octoprint (printserver similar to LaserWeb but used for 3d printer).
Its very useful to have the webcam pre-integrated on the page with all the control (up/down/left/right, laser power, weak laser on/off, start/stop/pause gcodeetc.).

Usually the webcam is used to setup (adjust and set the home point on the object) and safely monitor the engraving activity (emergency stop, pause, etc.).
Therefore all the contol button are needed beside the webcam live stream.

Implementation of control tab in octoprint:
[http://octoprint.org/assets/img/features/control-tab.png](url)

Screenshot from timelapse implementation in octoprint:
[http://octoprint.org/assets/img/features/timelapse-tab.png](url)

jorgerobles, 2017-09-22:
You mean this?

[image]

BTW. I know about Octoprint, but it uses other technology (Python) that currently offers low level system access, what is not posible currently on LW.

jorgerobles, 2017-09-22:
On the other side, LW offers OMR technology to make precise aligning of stock (https://github.com/LaserWeb/LaserWeb4/issues/354)

jorgerobles, 2017-10-21:
As #425, support for webcam URL is granted. Use a native streamer app on your machine to connect to, LW cannot afford an embedded streaming server performance, for now.

closed 2017-10-21 by jorgerobles