LaserWeb/LaserWeb4 #416: errors open eagle dxf files

madhouse-nl, 2017-09-25:
can you investigate the eagle7.7.0 output files  .dxf 

included in zip file.

I get this error massage when opening eagle dfx files:

Errors found in DMX_RGB_LED%N.sol.dxf please re-save in R12 ASCII format. Check console log and documentation for more details.
OK

[image]
[Nieuwe map (2).zip](https://github.com/LaserWeb/LaserWeb4/files/1329780/Nieuwe.map.2.zip)

jorgerobles, 2017-09-25:
Have you tried to resave as r12 dxf previously?

madhouse-nl, 2017-09-25:
In eagle there is no option for it.
from eagle file menu you can export in jpg, bmp or dxf but no options
john

jorgerobles, 2017-09-25:
Autocad is free download on Autodesk site. Lw only supports r12 dxf.

madhouse-nl, 2017-09-25:
Thanks, will try it.

closed 2017-09-25 by madhouse-nl