LaserWeb/LaserWeb4 #417: No supported firmware detected

jonbev, 2017-09-28:
I just updated to the latest code and I now get "No supported firmware detected"

App connected! (id=0)
INFO: Requesting Server Config 
INFO: Connecting to USB,/dev/ttyUSB0,115200
INFO: Connected to /dev/ttyUSB0 at 115200
No supported firmware detected. Closing port /dev/ttyUSB0

Is there a way to get more detailed output to see whats going on?

The board is xpro v3 with grbl 1.1.

It has been working fine until today.

cprezzi, 2017-09-28:
What OS are you running on?

cprezzi, 2017-09-28:
On my Raspi for example, I need to use /dev/ttyACM0 as the port.

jonbev, 2017-09-28:
I'm on Ubuntu.

Is there no verbose setting to see what is going on?

jorgerobles, 2017-09-28:
I've got running on ubuntu pretty well. Have you checked other ttyusb ports?

cprezzi, 2017-09-29:
You can activate logfile writing by adding "DEBUG_LEVEL=3" to a .env file in the installation folder, but I don't know if that works on Ubuntu. There would also be a "VERBOSE_LEVEL" (1-3) if you have the server console.
You could also use a terminal program and try if your suggested USB port is responding with the grbl welcome string. If that's the case, you could try to add "RESET_ON_CONNECT=1" to the .env file.

jonbev, 2017-09-29:
I'll give it a go. Thanks.

Going to assume this is user error and close this.

closed 2017-09-29 by jonbev