LaserWeb/LaserWeb4 #418: Support

tbfleming, 2017-09-28:
I was wondering why we're getting support questions here that belong in the community, so I checked the about page:

[image]

Too many people are going with the top choice. Maybe we should delete that link. The experienced users in the community could direct true bug reports here.

cojarbi, 2017-09-28:
Make sense to me

⁣Ariel Yahni​

cprezzi, 2017-09-28:
Ok for me. 
It would probably already help to add "- not for support questions!" behind the "- developer community text.

cprezzi, 2017-09-28:
Instead of the Github link we should add a link to the website https://cncpro.yurl.ch (or https://laserweb.yurl.ch) for detailed information and links to everywhere.