LaserWeb/LaserWeb4 #419: LW4 Files Quarantined By McAfee - Have Lost LW4

Bama-Breeze, 2017-10-01:
Attempted to download the most recent operating software for LW4 this evening.  It would run on a Windows 10 (64bit operating system) to control a C3D.  Previous version of LW4 downloaded without issue.

This time my McAfee quarantined the files and would not permit executing the installation.  I uninstalled the newest edition of LW4 and experienced the same circumstances with the previous edition of LW4 that had been installed successfully in July.  Consequently I have lost all copies of LW4.

Thank you for any guidence.

Attached are photos of the alert and error messages.
[image]

jorgerobles, 2017-10-01:
This issue have arised from time to time. The fact is an AV issue, I use avast, and no complains.

closed 2017-10-01 by jorgerobles

Bama-Breeze, 2017-10-01:
Thank you Jorge.  Seems all I tried to accomplish yesterday was snakebit.  My best sir. 

  LW4 (#419)