LaserWeb/LaserWeb4 #425: Provide Support for Raspberry Pi webcam

scottmil, 2017-10-12:
I am running LaserWeb4 on a Raspberry Pi 3 with the stock 8MP webcam.   I am running Raspbian Stretch version 4.9.41-v7+.

However, I cannot configure the Raspberry Pi camera in the LaserWeb4 settings.  The video resolution dropdown has no results.  (Screenshot attached)

Can you provide support for the Raspberry Pi webcam in a future release?

Thanks

[Laserweb-Workspace.json.txt](https://github.com/LaserWeb/LaserWeb4/files/1381042/Laserweb-Workspace.json.txt)
[image]

cprezzi, 2017-10-13:
We do not recommend running the frontend on a Raspberry Pi (too slow + needs Chome browser to work properly). You can run the server on the raspi, but you should use Win PC or Mac for the frontend.

If doing so, you need to stream the camera video via network to the frontend.

scottmil, 2017-10-13:
Thanks, Claudio.

I am running the LaserWeb4 server on a Raspi 3 and using my Mac Chrome browser as the front end, and it seems to run well.

I also have MJPEG-streamer successfully running on the same Raspi, whose video I stream independently to my frontend.  However, I was hoping I could better integrate the webcam video with the LaserWeb4 interface.

FYI: I run Octoprint on a Raspi with webcam for my 3D printer, and I was hoping to have the same web control functionality using LaserWeb4 with my laser engraver.

Again, thanks for all of your dedicated work on LaserWeb4.

-Scott

cprezzi, 2017-10-13:
@jorgerobles I thought we do support configuring a webcam via MJPEG streaming over IP? Don't we?

jorgerobles, 2017-10-13:
Nope. That's a pending feature :) I will take a shot next week.

jorgerobles, 2017-10-15:
@cprezzi, are you working on it on MPG branch? or is a different Mpeg?

cprezzi, 2017-10-16:
MPG stands for Motion Pulse Generator, which is a machine remote (pendant). In the MPG branch I am implementing an interface to the XHC-HB04 MPG (see http://www.ebay.com/itm/2017-Wireless-Mach3-MPG-Pendant-Handwheel-for-CNC-Mac-Mach-3-4-axis-Wholesale-/140918851970).

jorgerobles, 2017-10-16:
> Handwheel-for-CNC-Mac-Mach-3-4-axis-Wholesale-/140918851970).

Ok :)

closed 2017-10-21 by jorgerobles