LaserWeb/LaserWeb4 #426: translation

clemclemmm, 2017-10-15:
Hello world, 

This software is amazing, 
i want to contribute at my little level. 
But i'm not a programmer. 
But i know laser cutting and speaking french, 
so i can translate to french, 
if there is a table of words :-) 
(and i am building a laser cutter,
so i will be happy to have the french version when i will need it to run XD)
And i have lots of free hospital time, 
so it's perfect ^^

there is also other people wanting to help on it.
for other languages.

jorgerobles, 2017-10-15:
Possible implementation thru

https://github.com/i18next/react-i18next

jorgerobles, 2017-10-16:
Webpack https://www.npmjs.com/package/ya-i18next-webpack-plugin locales generator

tbfleming, 2017-10-17:
Costs to doing this:
* Most strings throughout the entire codebase will have to be wrapped. This is very error prone, especially if ones that shouldn't get wrapped end up wrapped accidentally. (e.g. strings which end up in the json format, strings which generate gcode, strings in messages to server, etc.)
* Most translation won't work by simple string replacement. Different languages have different word orderings. This means replacing a lot of "some text " + a + " more text " + b with substitution strings: format(translate("some text {1} more text {2}), a, b). Also error prone.
* We use a *ton* of libraries. Any strings we display which come from these will be untranslatable.
* Runtime performance: string constants turn into string lookups

jorgerobles, 2017-10-19:
Yep, but how deep will impact on software penetration?

tbfleming, 2017-10-19:
I have no way to predict that :(

SebCanet, 2022-10-04:
Hello,
sorry to open again this issue, few years later...but actually for STEM teachers, I'm looking for a translatable software, to change our old software (not open source, for old electronic controller, etc).
So potentially hundred of STEM teachers (pupills in fact :-D ) in France could use your software but only if translated.
Is there any modification about that?
Many thanks.

harlock999, 2022-10-04:
Bonjour!
From a development perspective, the wish list is starting to grow. Multi-language would likely be part of this wish list, but right now the short term focus is maintaining baseline usability as OS keep evolving and versions of dependencies evolve.

It would be possible to have a fork and hard-code translate but then I'm not sure how all future changes to the main software could easily ported to the fork. The original discussion from 2017 sums up the challenges pretty well, although I'm not so concerned about processing power now 5 years later.

What kind of machine are you looking to upgrade? Quel genre de machine pensez-vous mettre à jour?

SebCanet, 2022-10-04:
As I developp my own softwares, I understand that changing everything to insert a feature that was not in original core is a long work. The best is not hard coding translation but benefits of multi languages feature.

A very huge numbers of school in France ("collège" for us, aged 11 to 14) have Charly4U, CharlyRobot CNC [https://www.mecanumeric.fr/5-4/la-societe/charlyrobot-une-marque-de-mecanumeric/](https://www.mecanumeric.fr/5-4/la-societe/charlyrobot-une-marque-de-mecanumeric/). But a lot were bought many years ago, and electronic needs to be changed. So we want to propose an arduino kit grbl + free software, something easy for kids and open source to teach them this kind of numeric practice.

harlock999, 2022-10-04:
Since you have a very specific market, it would make sense to have a forked version to support these machines in the language of your liking. Eventually LaserWeb will mature enough that it will support multilanguage, but as things are in the open source world, timeline is unknown unless someone directly contributes or funds the feature.